zephyrproject-rtos/zephyr #72245: Weird console behaviour at boot on NRF52840 with CONFIG_LOG and CONFIG_SHELL

Salamandar, 2024-05-02:
I stumbled on a weird behaviour when testing Zephyr on nRF82540DK (supported by zephyr).
I also tested on Nucleo-L476RG and everything works as expected on that platform.

It appears with this config (commenting one or the other makes the bug disappear):
```
CONFIG_LOG=y
CONFIG_SHELL=y
```

At boot time, I'm expecting the banner `*** Booting Zephyr OS build v3.6.0 ***`. But it doesn't show. I got a `uart:~$` prompt instead, so console is working at that time.

Once in the main(), i got an even weirder behaviour. Here's my test code:
```
	led_on(led_gpio, 0);
	printk("before sleep\n");

	k_msleep(100);
	led_on(led_gpio, 1);
	printk("after 100ms sleep\n");

	k_msleep(1000);
	led_on(led_gpio, 2);
	printk("after sleep\n");
```

* `before sleep` is NEVER shown
* `after 100ms sleep` and `after sleep` messages are shown AT THE SAME TIME after 1.1s.
* LEDs are turned on at the expected "timestamps" though.

Here's my minimal reproducing project : https://github.com/Salamandar/zephyr-console-bug

When debugging, I tried to turn on `CONFIG_BOOT_DELAY=1000` but it doesn't change the behaviour.
BUT ! If i tinker in `zephyr/kernel/banner.c` and add `k_usleep(1);` everything works again as expected !? (with CONFIG_BOOT_DELAY commented out)

So it looks like it's a buffering issue and the k_usleep gives the opportunity to the buffer to initialize ?

**Expected behavior**
Banner and messages to be printed at the expected "moments".

**Environment (please complete the following information):**
 - Linux Manjaro
 - Zephyr SDK from the AUR
 - Zephyr tag v3.6.0

carlescufi, 2024-05-02:
What's your log backend in the `build/zephyr/.config` file? you can't have both the log and the shell competing for the UART, so you need to set `CONFIG_SHELL_LOG_BACKEND=y`

carlescufi, 2024-05-02:
Also, if you are using `printk`, then you need to ensure that `CONFIG_LOG_PRINTK=y`, otherwise printk and logging will also be fighting for the console

Salamandar, 2024-05-03:
Nice catch @carlescufi . I’m using `LOG_BACKEND_RTT=y`. If I disable that, everything works as expected !
What’s the consequence of this backend ?

carlescufi, 2024-05-03:
> Nice catch @carlescufi . I’m using `LOG_BACKEND_RTT=y`. If I disable that, everything works as expected ! What’s the consequence of this backend ?

Well, then logging will go over RTT, which means to see its output you need an RTT viewer from Segger, the UART will not contain the log output.

closed 2024-05-03 by carlescufi

Salamandar, 2024-05-03:
What I meant was : why selecting this backend affects the way early print messages are (not) displayed ?

carlescufi, 2024-05-06:
> What I meant was : why selecting this backend affects the way early print messages are (not) displayed ? And why calling k_usleep(1) in banner.c fixes the issue ? :(

This is possibly a race condition, but I am not sure. Pinging @nordic-krch to see if he can throw some light at this.

nordic-krch, 2024-05-07:
If you have more than one logging backend enabled then when at least one backend is ready messages are processed. At early stage shell is not yet ready (thread did not started) but RTT is ready (even though RTT on host does not capture the data). In that case, first messages goes to RTT (and they are lost). Once shell starts it also starts to receive logging messages. If you disable RTT then logging does not process messages until at least one backend is ready so messages are not lost and all are visible on shell.

Salamandar, 2024-05-13:
Thank you, that is a very understandable explanation.
I'll leave other maintainers to decide if that is a real issue interesting to fix.