zephyrproject-rtos/zephyr #72250: openamp_rsc_table example with MBOX: bus fault on TI SK-AM62 M4

aleeraser, 2024-05-02:
I'm using a TI SK-AM62 board running Linux on the main A53 and Zephyr on the M4 core. The Linux image is the [stock image](https://dr-download.ti.com/software-development/software-development-kit-sdk/MD-BDCgfEXHLk/09.01.00.08/tisdk-default-image-am62xx-evm.wic.xz) provided by TI.

I've been trying to adapt the `openamp_rsc_table` example for using the newer MBOX subsystem.
The MBOX driver is provided by https://github.com/zephyrproject-rtos/zephyr/pull/71241.

Here's the implementation I'm working on: https://github.com/aleeraser/zephyr/tree/feat/ti-mailbox, in particular [this commit](https://github.com/aleeraser/zephyr/commit/4bda969cb4b65efe8e692f1f6e014f3a668c6852) contains the changes for the example.

According to [TI's documentation](https://software-dl.ti.com/processor-sdk-linux/esd/AM62X/latest/exports/docs/linux/Foundational_Components_IPC62x.html#dma-memory-carveouts), the resource table is correctly loaded at the beginning of the M4 at `0x9cc00000`, while the IPC virtio and vring buffers must be located at `0x9cb00000`.

I have therefore created the following device tree overlay:

```
 / {
	chosen {
		zephyr,ipc_shm = &ipc_shm;
	};

	ipc_shm: memory@9cb00000 {
		compatible = "mmio-sram";
		reg = <0x9cb00000 DT_SIZE_M(1)>;
	};

	mbox-consumer {
		compatible = "vnd,mbox-consumer";
		mboxes = <&mbox0 0>, <&mbox0 1>;
		mbox-names = "tx", "rx";
	};
};
```

When running the example, I have the following Bus Fault / Precise data bus error:

```
[00:00:00.000,000] <err> os: ***** BUS FAULT *****
[00:00:00.000,000] <err> os:   Precise data bus error
[00:00:00.000,000] <err> os:   BFAR Address: 0x9cc0002d
[00:00:00.000,000] <err> os: r0/a1:  0x00000048  r1/a2:  0x000000ff  r2/a3:  0x9cc00014
[00:00:00.000,000] <err> os: r3/a4:  0x000077b8 r12/ip:  0x00000000 r14/lr:  0x000006e3
[00:00:00.000,000] <err> os:  xpsr:  0x01000000
[00:00:00.000,000] <err> os: Faulting instruction address (r15/pc): 0x00002bde
[00:00:00.000,000] <err> os: >>> ZEPHYR FATAL ERROR 25: Unknown error on CPU 0
[00:00:00.000,000] <err> os: Current thread: 0x75c0 (main)
[00:00:00.000,000] <err> os: Halting system
```

THe BFAR address (`0x9cc0002d`) should be within the resource table, and in fact `addr2line 0x00002bde` blames [this](https://github.com/zephyrproject-rtos/open-amp/blob/da78aea63159771956fe0c9263f2e6985b66e9d5/open-amp/lib/remoteproc/remoteproc_virtio.c) as the faulting instruction.

aleeraser, 2024-05-02:
@vaishnavachath any idea or suggestion to share?

vaishnavachath, 2024-05-07:
@glneo I believe you have got this working, do you have any pointers regarding the issue?

aleeraser, 2024-05-08:
> Yes, here is my working branch with `openamp_rsc_table` example running on M4 with Vaishnav's MBOX driver: https://github.com/glneo/zephyr/tree/openamp-am62x-sk-m4
> 
> The change you need to fix your data fault is this one: [4a04044](https://github.com/zephyrproject-rtos/zephyr/commit/4a04044155839f978b281cd3b39dd9e9311edf6e)

Thank you very much!

> The default RAT configuration is a bit of a mess right now, I'll send this patch upstream after a couple other bug fixes I've sent get through the pipeline.

I've seen your other patches. I agree this is definitely needed as well.

> Our solutions seem similar, we should be able to work together to find a clean way to update the current IPM based examples to the newer MBOX (maybe have it selected based on what is found in DT).

Unfortunately I'll be unavailable until the end of the month. Depending on the status of this when I'll be back, I'll be happy to join you if needed.