zephyrproject-rtos/zephyr #72260: Bluetooth: Host RX buffer allocation not optimal

cvinayak, 2024-05-03:
**Describe the bug**
Bluetooth Host Buffers are allocated for maximum count and maximum buffer size of the ACL RX, or  EVT RX or ISO RX counts and sizes.

https://github.com/zephyrproject-rtos/zephyr/blob/15fa7f79da1942d89a460112921a2b5c2e24c7b5/include/zephyr/bluetooth/buf.h#L88-L95

Typically, each of the buffer sizes and buffer counts are very much varying in the ranges, but a firmware is being built for maximum RAM allocation, and low RAM utilization.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -GNinja -DBOARD=nrf52833dk_nrf52820 -DEXTRA_CONF_FILE=overlay-bt_ll_sw_split.conf -DCONFIG_BT_ISO_RX_BUF_COUNT=1 -DCONFIG_BT_ISO_RX_MTU=251 ../../samples/bluetooth/iso_broadcast
3. ninja
4. See size
5. > Memory region         Used Size  Region Size  %age Used
           FLASH:      117096 B       256 KB     44.67%
             RAM:       24934 B        32 KB     76.09%
        IDT_LIST:          0 GB        32 KB      0.00%
6. <remove build files> 
7. Build with 10 bytes less RX MTU
8. cmake -GNinja -DBOARD=nrf52833dk_nrf52820 -DEXTRA_CONF_FILE=overlay-bt_ll_sw_split.conf -DCONFIG_BT_ISO_RX_BUF_COUNT=1 -DCONFIG_BT_ISO_RX_MTU=241 ../../samples/bluetooth/iso_broadcast
9. ninja
10. See size
11. > Memory region         Used Size  Region Size  %age Used
           FLASH:      117096 B       256 KB     44.67%
             RAM:       24834 B        32 KB     75.79%
        IDT_LIST:          0 GB        32 KB      0.00%
12. The difference is 10 times because in the sample, `CONFIG_BT_BUF_EVT_RX_COUNT=10`

**Expected behavior**
RAM usage should optimally reflect the Kconfig counts and size configured.

**Impact**
Showstopper for resource constraint devices.

**Logs and console output**
See description above for memory usage.

**Environment (please complete the following information):**
 - OS: MacOS
 - Toolchain: gnuarmemb (/usr/local/gcc-arm-none-eabi-9-2020-q2-update)
 - Commit SHA or Version used: 43651979fc3

**Additional context**
https://discord.com/channels/720317445772017664/1235629095526858762

jori-nordic, 2024-07-04:
I don't think this is a bug. It's the way the stack allocates memory due to the current HCI architecture.
I still agree we should optimize it though.

Thalley, 2024-11-20:
`BT_BUF_ISO_RX_SIZE` is defined incorrectly here. If this is for HCI, then it should never be above 255, but it is currently defined to just be `CONFIG_BT_ISO_RX_MTU` which can be `>>> 255`. 

We should have a separate Kconfig for ISO HCI buffer sizes and ISO SDU sizes.

Thalley, 2024-11-20:
Not sure what the solution here would be. We can limit `BT_BUF_ISO_RX_SIZE` to 255, but we still need X events of size Y, Z events of size K, etc. 

Currently we use MAX for all them, not sure why. It doesn't allow us to allocate the right amount of buffers (for example if `CONFIG_BT_BUF_EVT_RX_COUNT` is 10 and `CONFIG_BT_BUF_ACL_RX_COUNT` is 5, and we have allocated 5 buffers for ACL, then we only have 5, and not 10 left for HCI events.