zephyrproject-rtos/zephyr #72269: doc: bad list formatting in 'samples/subsys/fs/format/README'

Vyjill, 2024-05-03:
Bad formatting for a list in the first paragraph :

```md
This sample shows how to format different storage devices for different file systems. There are 2 scenarios prepared for this sample: 
* littleFS on flash device
* FAT file system on RAM disk
```

is most likely missing a line feed

```md
This sample shows how to format different storage devices for different file systems. There are 2 scenarios prepared for this sample: 

* littleFS on flash device
* FAT file system on RAM disk
```

**Environment**

N/A

* Page: `samples/subsys/fs/format/README`
* Version: main
* SHA-1: 8dc3f856229ce083c956aa301c31a23e65bd8cd8

kartben, 2024-05-03:
@Vyjill thanks for the report! If you could send a pull request fixing this, it would be much appreciated!

de-nordic, 2024-05-03:
[heart]         Ermel, Dominik reacted to your message:

PR #73538 (2024-05-30): samples: Fix list formatting in README
Added missing newline.

Fixes: #72269
commit 4ae207cc (2024-05-30): samples: Fix list formatting in README

Added missing newline.

Fixes: #72269
closed 2024-05-30 by dleach02