zephyrproject-rtos/zephyr #72275: <err> net_if: There is no network interface to work with! [ESP32-S3-Devkit] Zephyr 3.6.0

rvalvekar, 2024-05-03:
I have been trying to run the sample/net/wifi example on ESP32-S3 Devkit Board. 
The Build for the program is successfully being built. The Flashing process is also executed without any issues. 
When i run the west espressif monitor i see the following output :

west espressif monitor
Serial port /dev/ttyS3
Connecting.....
Detecting chip type... ESP32-S3
--- idf_monitor on /dev/ttyS3 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fca0000,len:0x2918
load:0x40384000,len:0x400
load:0x40384400,len:0x6ea8
load:0x4038d000,len:0x2cdc
entry 0x40385f9c
I (55) boot: MCUboot 2nd stage bootloader
I (56) boot: chip revision: v0.2
I (56) boot.esp32s3: Boot SPI Speed : 40MHz
I (56) boot.esp32s3: SPI Mode       : DIO
I (58) boot.esp32s3: SPI Flash Size : 8MB
I (62) boot: Enabling RNG early entropy source...
I (78) spi_flash: detected chip: winbond
I (78) spi_flash: flash io: dio
W (78) spi_flash: Detected size(16384k) larger than the size in the binary image header(8192k). Using the size in the binary image header.
[esp32s3] [INF] Image index: 0, Swap type: none
[esp32s3] [INF] Loading image 0 - slot 0 from flash, area id: 1
[esp32s3] [INF] Application start=403791B4h
[esp32s3] [INF] DRAM segment: paddr=000070E0h, vaddr=3FC8B0B0h, size=00F50h (  3920) load
[esp32s3] [INF] IRAM segment: paddr=00000040h, vaddr=40374000h, size=070A0h ( 28832) load
[esp32s3] [INF] DROM segment: paddr=00030000h, vaddr=3C030000h, size=019F0h (  6640) map
[esp32s3] [INF] IROM segment: paddr=00020000h, vaddr=42000000h, size=09436h ( 37942) map
[00:00:00.153,000] <err> net_if: There is no network interface to work with!
*** Booting Zephyr OS build v3.6.0-2202-ge3549278952e ***

The Error - There is no network interface to work with is where I am stuck. Other issues mention of a example wifi_station, but i did not find this example in the zephyr repo. 

Any help or direction how this could be resolved, would be highly appreciated. 

Thank you ! 
Rohan.

kartben, 2024-05-03:
You've installed Espressif's binary blobs, yes?

> Espressif HAL requires WiFi and Bluetooth binary blobs in order work. Run the command below to retrieve those files.
> `west blobs fetch hal_espressif`

rvalvekar, 2024-05-03:
> You've installed Espressif's binary blobs, yes?
> 
> > Espressif HAL requires WiFi and Bluetooth binary blobs in order work. Run the command below to retrieve those files.
> > `west blobs fetch hal_espressif`

Yes, i have installed binary blobs, and then executed west update. 
But still this error is seen. 

Thanks !

sylvioalves, 2024-05-03:
@rvalvekar, which esp32s3 module? Please, take a look at `samples/net/wifi/boards` and make sure you have the necessary overlay files, which enabled the wifi interface.

rvalvekar, 2024-05-03:
> @rvalvekar, which esp32s3 module? Please, take a look at `samples/net/wifi/boards` and make sure you have the necessary overlay/conf files, which enables the wifi interface.

@sylvioalves - I am using ESP32-S3-WROOM-1 N16R8. I was able to build couple of other codes, without any issues. I do have conf and overlay files added.

sylvioalves, 2024-05-03:
Is it devkitc or devkitm?

rvalvekar, 2024-05-03:
@sylvioalves - It is ESP32 devkitc,

sylvioalves, 2024-05-03:
This PR #72286 fix it. Anyway, you are missing the .conf and .overlay for esp32s3_devkitc in `boards` folders. Have you created it properly?

rvalvekar, 2024-05-03:
@sylvioalves - Yes ihave created it properly. I will check the fix, and will update.
Thanks for your support.

sylvioalves, 2024-05-03:
> @sylvioalves - Yes ihave created it properly. I will check the fix, and will update. Thanks for your support.

OK, let me also check here.

sylvioalves, 2024-05-03:
> @sylvioalves - Once the fix is implemented, would i just execute "west update" to get all the new changes that you have mentioned in PR #72286 ?
Pull latest master and then `west update`.

I've tested the devkitc and it worked. Added 'esp32s3_devkitc_procpu.overlay' and 'esp32s3_devkitc_procpu.conf' into `boards` folder.

rvalvekar, 2024-05-03:
@sylvioalves - Alright. I will check and update on this thread. 
Again thanks for the quick support.

rvalvekar, 2024-05-06:
@sylvioalves - Any solution for overflow in the iram_seg ? I checked a similar issue, and tried the solution of changing the size, but it has not helped. 

It now takes the board overlay and configurations without giving me an error while building, but this is new. 

below is the error output while building - 

[224/224] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
     mcuboot_hdr:          32 B         32 B    100.00%
        metadata:          32 B         32 B    100.00%
           FLASH:      137672 B    8388544 B      1.64%
     iram0_0_seg:       29332 B       299 KB      9.58%
     dram0_0_seg:       93984 B       283 KB     32.43%
     irom0_0_seg:       37610 B         8 MB      0.45%
     drom0_0_seg:       72200 B         8 MB      0.86%
    rtc_iram_seg:          0 GB         8 KB      0.00%
    rtc_data_seg:          0 GB         8 KB      0.00%
    rtc_slow_seg:          0 GB         8 KB      0.00%
        IDT_LIST:          0 GB         8 KB      0.00%
Generating files from /home/rohan/zephyrProjects/esp32s3WIFI/build/esp32s3WIFI/zephyr/zephyr.elf for board: esp32s3_devkitc
[11/16] Performing build step for 'mcuboot'
[1/239] Preparing syscall dependency handling

[4/239] Generating include/generated/version.h
-- Zephyr version: 3.6.99 (/home/rohan/zephyrproject/zephyr), build: v3.6.0-3413-g49d6b6bc0e36
[234/239] Linking C executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /home/rohan/zephyrProjects/esp32s3WIFI/build/mcuboot/zephyr/zephyr_pre0.map 
: && ccache /home/rohan/zephyr-sdk-0.16.5-1/xtensa-espressif_esp32s3_zephyr-elf/bin/xtensa-espressif_esp32s3_zephyr-elf-gcc  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/xtensa/core/offsets/offsets.c.obj  -fuse-ld=bfd  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/home/rohan/zephyrProjects/esp32s3WIFI/build/mcuboot/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/xtensa/core/libarch__xtensa__core.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/flash/libdrivers__flash.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  modules/mcuboot/libmcuboot_util.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/home/rohan/zephyr-sdk-0.16.5-1/xtensa-espressif_esp32s3_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32s3_zephyr-elf/12.2.0"  -L/home/rohan/zephyrProjects/esp32s3WIFI/build/mcuboot/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -lgcc  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie  -T/home/rohan/zephyrproject/modules/hal/espressif/zephyr/esp32s3/src/linker/esp32s3.rom.alias.ld  -T/home/rohan/zephyrproject/modules/hal/espressif/zephyr/esp32s3/../../components/esp_rom/esp32s3/ld/esp32s3.rom.ld  -T/home/rohan/zephyrproject/modules/hal/espressif/zephyr/esp32s3/../../components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld  -T/home/rohan/zephyrproject/modules/hal/espressif/zephyr/esp32s3/../../components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld  -T/home/rohan/zephyrproject/modules/hal/espressif/zephyr/esp32s3/../../components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld  -T/home/rohan/zephyrproject/modules/hal/espressif/zephyr/esp32s3/../../components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld  -T/home/rohan/zephyrproject/modules/hal/espressif/zephyr/esp32s3/../../components/soc/esp32s3/ld/esp32s3.peripherals.ld  -Wl,--wrap=longjmp && cd /home/rohan/zephyrProjects/esp32s3WIFI/build/mcuboot/zephyr && /usr/bin/cmake -E true
/home/rohan/zephyr-sdk-0.16.5-1/xtensa-espressif_esp32s3_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32s3_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32s3_zephyr-elf/bin/ld.bfd: zephyr/zephyr_pre0.elf section `.iram0.text' will not fit in region `iram_seg'
/home/rohan/zephyr-sdk-0.16.5-1/xtensa-espressif_esp32s3_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32s3_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32s3_zephyr-elf/bin/ld.bfd: section .iram0.loader_text LMA [00000000403bec00,00000000403c048f] overlaps section .iram0.text LMA [00000000403b6000,00000000403c0113]
/home/rohan/zephyr-sdk-0.16.5-1/xtensa-espressif_esp32s3_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32s3_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32s3_zephyr-elf/bin/ld.bfd: region `iram_seg' overflowed by 5396 bytes
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
[13/16] Completed 'esp32s3WIFI'
FAILED: images/bootloader/mcuboot-prefix/src/mcuboot-stamp/mcuboot-build /home/rohan/zephyrProjects/esp32s3WIFI/build/images/bootloader/mcuboot-prefix/src/mcuboot-stamp/mcuboot-build 
cd /home/rohan/zephyrProjects/esp32s3WIFI/build/mcuboot && /usr/bin/cmake --build .
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/rohan/zephyrProjects/esp32s3WIFI/build

sylvioalves, 2024-05-06:
@rvalvekar, reason is that this model is missing proper board configuration in mcuboot port. This fix it: 
https://github.com/mcu-tools/mcuboot/pull/1953  

You can manually create this file to overcome current build error.

rvalvekar, 2024-05-08:
@sylvioalves 

This is a new topic regarding same WIFI and drivers - undefined reference to `wifi_mgmt_connect'

With the help of your new overlays, it was possible to build the binaries, but still it is not detecting interface. So I am trying something else, and i encountered this error. 

FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /home/rohan/zephyrProjects/esp32s3WIFI/build/esp32s3WIFI/zephyr/zephyr_pre0.map 
: && ccache /home/rohan/zephyr-sdk-0.16.5-1/xtensa-espressif_esp32s3_zephyr-elf/bin/xtensa-espressif_esp32s3_zephyr-elf-gcc  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/xtensa/core/offsets/offsets.c.obj  -fuse-ld=bfd  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/home/rohan/zephyrProjects/esp32s3WIFI/build/esp32s3WIFI/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/xtensa/core/libarch__xtensa__core.a  zephyr/lib/libc/picolibc/liblib__libc__picolibc.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/subsys/random/libsubsys__random.a  zephyr/subsys/net/libsubsys__net.a  zephyr/subsys/net/l2/ethernet/libsubsys__net__l2__ethernet.a  zephyr/subsys/net/l2/wifi/libsubsys__net__l2__wifi.a  zephyr/subsys/net/ip/libsubsys__net__ip.a  zephyr/subsys/net/lib/dhcpv4/libsubsys__net__lib__dhcpv4.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/entropy/libdrivers__entropy.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  zephyr/drivers/wifi/libdrivers__wifi.a  modules/mbedtls/libmbedTLSBase.a  modules/mbedtls/libmbedTLSCrypto.a  modules/mbedtls/libmbedTLSX509.a  modules/mbedtls/libmodules__mbedtls.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/home/rohan/zephyr-sdk-0.16.5-1/xtensa-espressif_esp32s3_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32s3_zephyr-elf/12.2.0"  -L/home/rohan/zephyrProjects/esp32s3WIFI/build/esp32s3WIFI/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  modules/mbedtls/libmbedTLSBase.a  modules/mbedtls/libmbedTLSCrypto.a  modules/mbedtls/libmbedTLSX509.a  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie  -DPICOLIBC_LONG_LONG_PRINTF_SCANF  -T/home/rohan/zephyrproject/modules/hal/espressif/zephyr/esp32s3/src/linker/esp32s3.rom.alias.ld  -T/home/rohan/zephyrproject/modules/hal/espressif/zephyr/esp32s3/../../components/esp_rom/esp32s3/ld/esp32s3.rom.ld  -T/home/rohan/zephyrproject/modules/hal/espressif/zephyr/esp32s3/../../components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld  -T/home/rohan/zephyrproject/modules/hal/espressif/zephyr/esp32s3/../../components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld  -T/home/rohan/zephyrproject/modules/hal/espressif/zephyr/esp32s3/../../components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld  -T/home/rohan/zephyrproject/modules/hal/espressif/zephyr/esp32s3/../../components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld  -T/home/rohan/zephyrproject/modules/hal/espressif/zephyr/esp32s3/../../components/soc/esp32s3/ld/esp32s3.peripherals.ld  -Wl,--wrap=longjmp  -lnet80211  -lcore  -lpp  -lphy  -lgcc  -L/home/rohan/zephyrproject/modules/hal/espressif/zephyr/esp32s3/../blobs/lib/esp32s3  --specs=picolibc.specs  -lc  -lgcc && cd /home/rohan/zephyrProjects/esp32s3WIFI/build/esp32s3WIFI/zephyr && /usr/bin/cmake -E true
/home/rohan/zephyr-sdk-0.16.5-1/xtensa-espressif_esp32s3_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32s3_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32s3_zephyr-elf/bin/ld.bfd: app/libapp.a(main.c.obj):(.literal.main+0x14): undefined reference to `wifi_mgmt_connect'
/home/rohan/zephyr-sdk-0.16.5-1/xtensa-espressif_esp32s3_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32s3_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32s3_zephyr-elf/bin/ld.bfd: app/libapp.a(main.c.obj): in function `main':
/home/rohan/zephyrProjects/esp32s3WIFI/src/main.c:17: undefined reference to `wifi_mgmt_connect'

sylvioalves, 2024-07-10:
Fixed by https://github.com/mcu-tools/mcuboot/pull/1953

closed 2024-07-10 by sylvioalves

StefanHri, 2024-11-09:
I have the same problem with esp32c6 (ESP32-C6-WROOM-1 module). What exactly do I need to add the `boards` and `socs` folders?

sylvioalves, 2024-11-09:
@StefanHri ESP32-C6 WiFi support is not yet merged into main repository:  #81078

StefanHri, 2024-11-09:
When we can expect this to happen?

sylvioalves, 2024-11-09:
I guess in a couple of days. Next week probably.

kartben, 2024-11-10:
> I guess in a couple of days. Next week probably. 

Rather the week after next, i.e. after Zephyr 4.0 is out :)