zephyrproject-rtos/zephyr #72304: NVS Settings Deadlock

ian-678, 2024-05-03:
**Describe the bug**
Repeated resets (shortly after boot) and writing nvs settings to internal flash causes nvs_init to eventually return -EDEADLK. At this point nvs settings can no longer use the internal flash and erasing the flash is required to free it up. Reading out the flash when deadlock is present shows there are no empty sectors (0xff) in the nvs settings section (see attached).

If all sectors are closed shouldn't garbage collection run and clear it out? Or is the deadlock an unrecoverable state for NVS?

- nrf52840 on custom board
- NCS 1.9.2
- NCS Zephyr fork v2.7.99-ncs1-1
- Toolchain: Zephyr SDK 0.14.2
- ble and coap/openthread in use

**To Reproduce**
- Power-cycle the board every 30 seconds, leave overnight
- Unable to replicate as of yet on a nrf52840 dev kit

**Expected behavior**
Settings recovers itself at init if deadlock occurs.

**Impact**
All subsystems that use settings are unusable. Showstopper.

**Flash Dump**
Settings section is 0xF8000 with length 0x8000.
[flash_dump_settings.zip](https://github.com/zephyrproject-rtos/zephyr/files/15204690/flash_dump_settings.zip)

aescolar, 2024-05-07:
CC @PavelVPV @alxelax @de-nordic

de-nordic, 2024-05-07:
CC @Laczen

nashif, 2024-05-07:
This is not a zephyr upstream issue, please do not file issues of NCS upstream(zephyr), file those downstream (NCS).

closed 2024-05-07 by nashif

JtheSaw, 2026-02-12:
Okay thank you for your response

I also opened up an issue on Nordics Devzone for my problem:
https://devzone.nordicsemi.com/f/nordic-q-a/126968/nrf_nvmc--config-hangs-on-setting-nrf_nvmc_mode_write-in-zephyr-without-softdevice