zephyrproject-rtos/zephyr #72308: tests: lpuart: uart_async_api: build failure for mimxrt10xx series

hakehuang, 2024-05-04:
**Describe the bug**
Build failure

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 -  mimxrt10xx, mr_canhubk3, mimxrt11xx series
 - What have you tried to diagnose or workaround this issue?
 - NA
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 
```
9320907fdf1f6dc436c426d4d31f0a0f6d2d2e8f is the first bad commit
commit 9320907fdf1f6dc436c426d4d31f0a0f6d2d2e8f
Author: Declan Snyder <declan.snyder@nxp.com>
Date:   Wed Apr 24 19:37:48 2024 -0500

    drivers: uart_mcux_lpuart: Fix APIs used together
    
    Handle controller instances wanting to use different
    API instances, need to change init flow so that they dont
    interfere with each other, trick for now is returning
    either the interrupt driven or async apis as not enabled
    if the other has already been used on that controller.
    
    Previously, there was an issue where enabling CONFIG_UART_ASYNC_API
    would overwrite the init even if controller was meant to use
    interrupt driven (as kconfig is global to affect all the controllers)
    
    Signed-off-by: Declan Snyder <declan.snyder@nxp.com>

 drivers/serial/uart_mcux_lpuart.c | 193 ++++++++++++++++++++++++++++----------
 1 file changed, 143 insertions(+), 50 deletions(-)
```

**To Reproduce**
Steps to reproduce the behavior:

scripts/twister -p mimxrt10160_evk -T mimxrt1170_evk@B/mimxrt1176/cm7 --build-only

**Expected behavior**
Build pass

**Impact**
build CI

**Logs and console output**

```
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/drivers/serial/uart_mcux_lpuart.c:1030:12: error: 'mcux_lpuart_configure_async' defined but not used [-Werror=unused-function]
 1030 | static int mcux_lpuart_configure_async(const struct device *dev)
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
ninja: build stopped: subcommand failed.
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v3.6.0-3395-g1ae357850d5c

decsny, 2024-05-15:
@hakehuang i don't understand the reproduce step you gave, what sample did you run, and was it rt1060 or rt1170?

hakehuang, 2024-05-16:
> @hakehuang i don't understand the reproduce step you gave, what sample did you run, and was it rt1060 or rt1170?

@decsny  it is not  a run problem, it is a build issue, zephyr CI does not allow build warning, and in this driver there is an unused fucntion which will cause CI to warn, and the CI build flag is set to wran to error

nordic-krch, 2024-05-16:
any news here? I'm also getting CI failures: https://github.com/zephyrproject-rtos/zephyr/actions/runs/9108541914/job/25039692154?pr=71260

PR #72955 (2024-05-18): drivers: lpuart: Fix async configure
Fix async configure function not being called
when the irq driven api is not used, which
both causes a build warning and would not
work at runtime.

Fixes #72308
closed 2024-05-18 by jhedberg