zephyrproject-rtos/zephyr #72312: Incorrect data scaling in IST8310 magentic sensor driver

mluessi, 2024-05-04:
The driver for the IST8310 magnetometer scales the raw data by a factor of `1.0f / 1320` (i.e., 0.000757576) to convert it to Gauss. However, according to the [datasheet](https://tw.isentek.com/userfiles/files/IST8310Datasheet_3DMagneticSensors.pdf), the resolution of the raw data is 0.3 uT/LSB, which corresponds to 0.003 G/LSB. Therefore, the data should be scaled by `0.003f`, not `1.0f / 1320` .

Note that the IST8308 sensor from the same manufacturer has a resolution of 0.00075 G/LSB, so it is possible that the bug originates from there.

If someone can confirm that this is indeed a bug, I am happy to open a pull request for a fix.

MaureenHelm, 2024-05-06:
@PetervdPerk-NXP will you please take a look?

teburd, 2024-05-07:
I don't have this device, removing myself from assignee

mluessi, 2024-05-09:
@PetervdPerk-NXP I have a number of different magnetometers here and will do some tests. If the datasheet is incorrect, we should ask iSentek for clarification.

PetervdPerk-NXP, 2024-05-10:
So the datasheet mentions 330 LSB/Gauss but that's in 14-bit mode.
We're running in 16-bit mode which is 1320 LSB/Gauss (1320 = 330 * (2^16/2^14))
The website page of the IST8310 also mentions 1320 LSB/Gauss but the datasheet does not.
https://isentek.com/products_view.php?PID=10&sn=13

mluessi, 2024-05-10:
@PetervdPerk-NXP  thank you for looking into this more and providing the source of the 1320 LSB/Gauss scaling. 

 I compared it to a RM3100 magnetometer and it gives comparable values with the scaling  implemented in zephyr, which also confirms that it is correct and not a bug. I will close this issue.

closed 2024-05-10 by mluessi