zephyrproject-rtos/zephyr #72318: Possible bug in twister gtest harness

GeekOffTheStreet, 2024-05-04:
### Discussed in https://github.com/zephyrproject-rtos/zephyr/discussions/71635

<div type='discussions-op-text'>

<sup>Originally posted by **GeekOffTheStreet** April 17, 2024</sup>
I tried using Twister with the gtest harness and ran into issues with it detecting unit test execution.   The behavior for how it parses sample.yaml/testcase.yaml doesn't match what I would expect. I should not that I'm not using pigweed.  Consider this simple test case:

```C++
#include <gtest/gtest.h>
#include <gmock/gmock.h>
#include <stdio.h>
using namespace testing;

class HelloWorld : public Test {
public:
    HelloWorld() {}
    void SetUp() {}
    void TearDown() {}
};

TEST_F(HelloWorld, foo) {
    EXPECT_TRUE(true);
}
```

I build that for qemu using:
```
$ west build -p always -b qemu_leon3--build-dir zephyr-build .
$ /opt/zephyrproject/zephyr/scripts/twister  -vvv -p qemu_leon3 -T ./
```

If I  use the console harness with a sample.yaml resembling:
```yaml
sample:
  description: Hello World sample, the simplest Zephyr
  name: test hello world
common:
  tags: introduction
  platform_allow: qemu_leon3
  harness: console
  harness_config:
    type: multi_line
    ordered: false
    regex:
      - ".*OK\\s+\\]\\s+HelloWorld.foo.+"

tests:
  HelloWorld.foo:
    tags: introduction
```

The test works as expected (sample output):
```
DEBUG   - Determine test cases for test instance: qemu_leon3/HelloWorld.foo
DEBUG   - Determine test cases for test suite: HelloWorld.foo
DEBUG   - Test instance qemu_leon3/HelloWorld.foo already has 1 cases.
DEBUG   - run test: qemu_leon3/HelloWorld.foo
DEBUG   - Reset instance status from 'passed' to None before run.
DEBUG   - Spawning QEMUHandler Thread for qemu_leon3/HelloWorld.foo
DEBUG   - Running qemu_leon3/HelloWorld.foo (qemu)
DEBUG   - Spawning QEMUHandler Thread for qemu_leon3/HelloWorld.foo
DEBUG   - QEMU (232776): [==========] Running 1 test from 1 test case.
DEBUG   - QEMU (232776): [----------] Global test environment set-up.
DEBUG   - QEMU (232776): [----------] 1 test from HelloWorld
DEBUG   - QEMU (232776): [ RUN      ] HelloWorld.foo
DEBUG   - QEMU (232776): [       OK ] HelloWorld.foo (0 ms)
DEBUG   - HARNESS:Console:EXPECTED(1/1):'.*OK\s+\]\s+HelloWorld.foo.+'
DEBUG   - QEMU (232776): [----------] 1 test from HelloWorld (0 ms total)
DEBUG   - QEMU (232776): 
DEBUG   - QEMU (232776): [----------] Global test environment tear-down
DEBUG   - QEMU (232776): [==========] 1 test from 1 test case ran. (10 ms total)
DEBUG   - QEMU (232776): [  PASSED  ] 1 test.
DEBUG   - QEMU (232776): *** Booting Zephyr OS build zephyr-v3.5.0-2466-g654a74ecb7d6 ***
DEBUG   - QEMU (232776) complete (passed) after 2.050739288330078 seconds
DEBUG   - No timeout, return code from QEMU (None): 0
DEBUG   - return code from QEMU (None): 0
DEBUG   - run status: qemu_leon3/HelloWorld.foo passed
INFO    - 1/1 qemu_leon3                HelloWorld.foo                                     PASSED (qemu 2.051s)

INFO    - 1 test scenarios (1 test instances) selected, 0 configurations skipped (0 by static filter, 0 at runtime).
```

If I try to use the gtest harness:
```yaml
sample:
  description: Hello World sample, the simplest Zephyr
  name: test hello world
common:
  tags: introduction
  platform_allow: qemu_leon3
  harness: gtest
tests:
  HelloWorld.foo:
    tags: introduction
```

The runner tries to match `HelloWorld.foo.HelloWorld.foo` instead of just `HelloWorld.foo` and fails:
```
DEBUG   - Determine test cases for test instance: qemu_leon3/HelloWorld.foo
DEBUG   - Determine test cases for test suite: HelloWorld.foo
DEBUG   - Test instance qemu_leon3/HelloWorld.foo already has 1 cases.
DEBUG   - run test: qemu_leon3/HelloWorld.foo
DEBUG   - Reset instance status from 'passed' to None before run.
DEBUG   - Spawning QEMUHandler Thread for qemu_leon3/HelloWorld.foo
DEBUG   - Running qemu_leon3/HelloWorld.foo (qemu)
DEBUG   - Spawning QEMUHandler Thread for qemu_leon3/HelloWorld.foo
DEBUG   - QEMU (234555): [==========] Running 1 test from 1 test case.
DEBUG   - QEMU (234555): [----------] Global test environment set-up.
DEBUG   - QEMU (234555): [----------] 1 test from HelloWorld
DEBUG   - QEMU (234555): [ RUN      ] HelloWorld.foo
DEBUG   - Could not find a matching testcase for HelloWorld.foo.HelloWorld.foo
DEBUG   - QEMU (234555): [       OK ] HelloWorld.foo (0 ms)
Exception in thread qemu_leon3/HelloWorld.foo:
Traceback (most recent call last):
  File "/usr/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.10/threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "/opt/zephyrproject/zephyr/scripts/pylib/twister/twisterlib/handlers.py", line 902, in _thread
    harness.handle(line)
  File "/opt/zephyrproject/zephyr/scripts/pylib/twister/twisterlib/harness.py", line 573, in handle
    tc is not None and tc == self.tc
AssertionError: gTest error, mismatched tests. Expected HelloWorld.foo.HelloWorld.foo but got None
DEBUG   - return code from QEMU (None): 2
DEBUG   - run status: qemu_leon3/HelloWorld.foo failed
INFO    - 1/1 qemu_leon3                HelloWorld.foo                                      FAILED Timeout (qemu)
```

I didn't see an obvious way to alter the matching behavior.

@yperess 
</div>

yperess, 2024-05-04:
Thank you, I'll take a look this week.

vChavezB, 2024-06-16:
I am also experiencing a similar issue with twister. In my case I want to integrate CPPUTest so I had a look at the Gtest Harness. This brought me to this issue.

As far as I understand the problem is that each test case name obtained dynamically from the std output of the Harness (Gtest in this case), is created by the format `{test case id}.{test suite}.{test case}`. This is done specifically here:

https://github.com/zephyrproject-rtos/zephyr/blob/4d0a8c1e8b8316ec5071a3fcc98cd292d3bc4e6d/scripts/pylib/twister/twisterlib/harness.py#L582

I think that is the reason we get from harness handle from Gtest a created instance with name `HelloWorld.foo.HelloWorld.foo`.

What happen is the following:

1. Test suite `Hello world` with test case `Foo` runs and prints

    `[ RUN      ] HelloWorld.foo`

2. The [Test instance](https://github.com/zephyrproject-rtos/zephyr/blob/4d0a8c1e8b8316ec5071a3fcc98cd292d3bc4e6d/scripts/pylib/twister/twisterlib/testinstance.py) module does not find the instance and prints 

    `DEBUG   - Could not find a matching testcase for HelloWorld.foo.HelloWorld.foo`
    
    This happens due to the format I mentioned at the beginning. This in turn creates a new test instance called `HelloWorld.foo.HelloWorld.foo`
    
3. Test suite `Hello world` with test case `Foo` finishes and prints

    `[OK ] HelloWorld.foo (0 ms) `
    
4. The GTest Harness detects that a test finished here:

    https://github.com/zephyrproject-rtos/zephyr/blob/4d0a8c1e8b8316ec5071a3fcc98cd292d3bc4e6d/scripts/pylib/twister/twisterlib/harness.py#L613
    
5. Now here is the important part. The refex pattern [`TEST_PASS_PATTERN`](https://github.com/zephyrproject-rtos/zephyr/blob/4d0a8c1e8b8316ec5071a3fcc98cd292d3bc4e6d/scripts/pylib/twister/twisterlib/harness.py#L556C5-L556C23) I think is wrong. If I test it with regex101.com this is the result I get https://regex101.com/r/Uj6EYf/1

    [image]
    
    As you see the string representation of the test case will be `HelloWorld.foo.HelloWorld.foo (0 ms)` instead of the expected `HelloWorld.foo.HelloWorld.foo`. This in turn returns None 
    
    https://github.com/zephyrproject-rtos/zephyr/blob/4d0a8c1e8b8316ec5071a3fcc98cd292d3bc4e6d/scripts/pylib/twister/twisterlib/testinstance.py#L159-L163

    and makes the assertion fail.
    
    https://github.com/zephyrproject-rtos/zephyr/blob/4d0a8c1e8b8316ec5071a3fcc98cd292d3bc4e6d/scripts/pylib/twister/twisterlib/harness.py#L620-L623
    
## Possible solution

Change the Pass pattern (and similarly the fail pattern as well) to something as 

`.*\[       OK \] (?P<suite_name>\S+)\.(?P<test_name>\S+) \(\d+ ms\)`

[image]

https://regex101.com/r/NX5ul9/1

byteinthewild, 2024-07-30:
I recently stumbled on this issue because I'm seeing the same error 
`DEBUG   - Could not find a matching testcase for HelloWorld.foo.HelloWorld.foo`

It looks like the pull request resolved another issue related to test timing being at the end of the gtest output, but did not address the duplicated suite and test name in the search?  or am I missing something here?

My testcase.yaml looks like:

`
tests:
  HelloWorld.foo:
    tags: foo
`

yperess, 2024-07-31:
@byteinthewild I'm on vacation right now, but I'll try to take a look ASAP

byteinthewild, 2024-07-31:
Great, thank you very much, enjoy your vacation!

akrenz, 2024-08-11:
Applying the changes from here: https://github.com/zephyrproject-rtos/zephyr/pull/65430/files fixed the issue for Googletest 1.15.0.

Unfortunately the changes have not been merged into zephyr.

yperess, 2024-08-15:
Started working on this, should have it fixed by the end of the week.

PR #75228 (2024-07-03): twister: Fix gTest harness
The gTest harness asssumed that the lines end with the test name, but some gTest implementations include the test duration in the line. Update both the tests and regex to allow this and also avoid capturing characters into the `test_name` that cannot be valid test name chars.

Fixes #72318
commit 1b51740f (2024-07-03): twister: Fix gTest harness

The gTest harness asssumed that the lines end with the test name, but
some gTest implementations include the test duration in the line. Update
both the tests and regex to allow this and also avoid capturing
characters into the `test_name` that cannot be valid test name chars.

Fixes #72318
PR #77134 (2024-08-23): gtest: Add support for gtest 1.15.0
The gtest 1.15 has slightly different output from the original one that was used to set up the harness. With this change we support both output formats.

Fixes #72318
commit bfb893a0 (2024-08-23): gtest: Add support for gtest 1.15.0

The gtest 1.15 has slightly different output from the original one that
was used to set up the harness. With this change we support both output
formats.

Fixes #72318
closed 2024-08-23 by aescolar