zephyrproject-rtos/zephyr #72325: Build Failure in GSM Modem Sample

nmpluta, 2024-05-05:
**Describe the bug**
When building the [gsm_modem sample](https://docs.zephyrproject.org/latest/samples/net/gsm_modem/README.html) according to the documentation, the build fails with the following error:
```
/home/npluta/repositories/private/workspace-telemetry-system/zephyr/include/zephyr/device.h:91:41: error: '__device_dts_ord___ORD' undeclared here (not in a function); did you mean '__device_dts_ord_3'?
```
This error arises due to an undeclared symbol in the device.h header file.
- Target platforms tested: reel_board, frdm_k64f

**To Reproduce**
Steps to reproduce the behavior:
1. Build the sample with the provided below command:
```
west build -b reel_board samples/net/gsm_modem -- -DCONFIG_MODEM_GSM_APN=\"internet\"
```

**Expected behavior**
The `gsm_modem` sample should build successfully without any errors.

**Impact**
This issue prevents the successful building of the `gsm_modem` sample.

**Logs and console output**
```
zephyr (49d6b6b) [$!+] 
❯ west build -b reel_board samples/net/gsm_modem -- -DCONFIG_MODEM_GSM_APN=\"internet\"
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/npluta/repositories/private/workspace-telemetry-system/zephyr/samples/net/gsm_modem
-- CMake version: 3.27.7
-- Found Python3: /usr/bin/python3 (found suitable version "3.10.12", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/npluta/.cache/zephyr
-- Zephyr version: 3.6.99 (/home/npluta/repositories/private/workspace-telemetry-system/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: reel_board, Revision: 1, qualifiers: nrf52840
-- Found host-tools: zephyr 0.16.5 (/home/npluta/zephyr-sdk-0.16.5-1)
-- Found toolchain: zephyr 0.16.5 (/home/npluta/zephyr-sdk-0.16.5-1)
-- Found Dtc: /home/npluta/zephyr-sdk-0.16.5-1/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/npluta/repositories/private/workspace-telemetry-system/zephyr/boards/phytec/reel_board/reel_board.dts
-- Generated zephyr.dts: /home/npluta/repositories/private/workspace-telemetry-system/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/npluta/repositories/private/workspace-telemetry-system/zephyr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/npluta/repositories/private/workspace-telemetry-system/zephyr/build/zephyr/dts.cmake

warning: Deprecated symbol MODEM_GSM_PPP is enabled.

Parsing /home/npluta/repositories/private/workspace-telemetry-system/zephyr/Kconfig
Loaded configuration '/home/npluta/repositories/private/workspace-telemetry-system/zephyr/boards/phytec/reel_board/reel_board_defconfig'
Merged configuration '/home/npluta/repositories/private/workspace-telemetry-system/zephyr/samples/net/gsm_modem/prj.conf'
Merged configuration '/home/npluta/repositories/private/workspace-telemetry-system/zephyr/build/zephyr/misc/generated/extra_kconfig_options.conf'
Configuration saved to '/home/npluta/repositories/private/workspace-telemetry-system/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/npluta/repositories/private/workspace-telemetry-system/zephyr/build/zephyr/include/generated/autoconf.h'
-- Found GnuLd: /home/npluta/zephyr-sdk-0.16.5-1/arm-zephyr-eabi/arm-zephyr-eabi/bin/ld.bfd (found version "2.38") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/npluta/zephyr-sdk-0.16.5-1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Using ccache: /usr/bin/ccache
-- Configuring done (3.4s)
-- Generating done (0.0s)
-- Build files have been written to: /home/npluta/repositories/private/workspace-telemetry-system/zephyr/build
-- west build: building application
[1/244] Preparing syscall dependency handling

[3/244] Generating include/generated/version.h
-- Zephyr version: 3.6.99 (/home/npluta/repositories/private/workspace-telemetry-system/zephyr), build: v3.6.0-3413-g49d6b6bc0e36
[38/244] Building C object CMakeFiles/app.dir/src/main.c.obj
FAILED: CMakeFiles/app.dir/src/main.c.obj 
ccache /home/npluta/zephyr-sdk-0.16.5-1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DKERNEL -DK_HEAP_MEM_POOL_SIZE=0 -DNRF52840_XXAA -DPICOLIBC_DOUBLE_PRINTF_SCANF -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 -I/home/npluta/repositories/private/workspace-telemetry-system/zephyr/include -I/home/npluta/repositories/private/workspace-telemetry-system/zephyr/build/zephyr/include/generated -I/home/npluta/repositories/private/workspace-telemetry-system/zephyr/soc/nordic -I/home/npluta/repositories/private/workspace-telemetry-system/zephyr/lib/posix/options/getopt/. -I/home/npluta/repositories/private/workspace-telemetry-system/zephyr/soc/nordic/nrf52/. -I/home/npluta/repositories/private/workspace-telemetry-system/zephyr/soc/nordic/common/. -I/home/npluta/repositories/private/workspace-telemetry-system/zephyr/subsys/net/lib/dns/. -I/home/npluta/repositories/private/workspace-telemetry-system/zephyr/subsys/net/conn_mgr/. -I/home/npluta/repositories/private/workspace-telemetry-system/telemetry-system/include -I/home/npluta/repositories/private/workspace-telemetry-system/modules/hal/cmsis/CMSIS/Core/Include -I/home/npluta/repositories/private/workspace-telemetry-system/zephyr/modules/cmsis/. -I/home/npluta/repositories/private/workspace-telemetry-system/modules/hal/nordic/nrfx -I/home/npluta/repositories/private/workspace-telemetry-system/modules/hal/nordic/nrfx/drivers/include -I/home/npluta/repositories/private/workspace-telemetry-system/modules/hal/nordic/nrfx/mdk -I/home/npluta/repositories/private/workspace-telemetry-system/zephyr/modules/hal_nordic/nrfx/. -isystem /home/npluta/repositories/private/workspace-telemetry-system/zephyr/lib/libc/common/include -fno-strict-aliasing -Os -imacros /home/npluta/repositories/private/workspace-telemetry-system/zephyr/build/zephyr/include/generated/autoconf.h -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfp16-format=ieee -mtp=soft --sysroot=/home/npluta/zephyr-sdk-0.16.5-1/arm-zephyr-eabi/arm-zephyr-eabi -imacros /home/npluta/repositories/private/workspace-telemetry-system/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wdouble-promotion -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/home/npluta/repositories/private/workspace-telemetry-system/zephyr/samples/net/gsm_modem=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/npluta/repositories/private/workspace-telemetry-system/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/npluta/repositories/private/workspace-telemetry-system=WEST_TOPDIR -ffunction-sections -fdata-sections --specs=picolibc.specs -D_POSIX_THREADS -std=c99 -MD -MT CMakeFiles/app.dir/src/main.c.obj -MF CMakeFiles/app.dir/src/main.c.obj.d -o CMakeFiles/app.dir/src/main.c.obj -c /home/npluta/repositories/private/workspace-telemetry-system/zephyr/samples/net/gsm_modem/src/main.c
In file included from /home/npluta/repositories/private/workspace-telemetry-system/zephyr/include/zephyr/toolchain/gcc.h:98,
                 from /home/npluta/repositories/private/workspace-telemetry-system/zephyr/include/zephyr/toolchain.h:50,
                 from /home/npluta/repositories/private/workspace-telemetry-system/zephyr/include/zephyr/kernel_includes.h:23,
[... 44 lines cut ...]
      |                          ^
/home/npluta/repositories/private/workspace-telemetry-system/zephyr/include/zephyr/device.h:91:33: note: in expansion of macro '_CONCAT'
   91 | #define DEVICE_NAME_GET(dev_id) _CONCAT(__device_, dev_id)
      |                                 ^~~~~~~
/home/npluta/repositories/private/workspace-telemetry-system/zephyr/include/zephyr/device.h:228:37: note: in expansion of macro 'DEVICE_NAME_GET'
  228 | #define DEVICE_DT_NAME_GET(node_id) DEVICE_NAME_GET(Z_DEVICE_DT_DEV_ID(node_id))
      |                                     ^~~~~~~~~~~~~~~
/home/npluta/repositories/private/workspace-telemetry-system/zephyr/include/zephyr/device.h:245:34: note: in expansion of macro 'DEVICE_DT_NAME_GET'
[... cut ...]
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /home/npluta/.local/bin/cmake --build /home/npluta/repositories/private/workspace-telemetry-system/zephyr/build
```

**Environment (please complete the following information):**
- OS: Ubuntu 22.04.4 LTS
- Toolchain: Zephyr SDK 0.16.5
- Commit SHA or Version used: 49d6b6bc0e36247157af2ad03ffa96aff926fd73 (Zephyr v3.6.99)

**Additional context**
The sample was tested for the `frdm_k64f` target as well, but it's failing with the same error.

henrikbrixandersen, 2024-05-05:
Neither the `reel_board` nor the `frdm_k64f` contains a modem. How can you expect to use a modem-specific sample on any of those board?

nmpluta, 2024-05-05:
Hi @henrikbrixandersen, 

according to the documentation this sample supports _an external modem like the SIMCOM card_:
[image]

I assumed that it would work out of the box with the target from the doc (`reel_board`) and connected external modem, which failed due to not configured modem. Upon investigation, I found that the modem configuration was missing, which caused the issue.  Creating the following file: `reel_board.overlay`  fixes the issue.
```
&arduino_serial {
	status = "okay";
	current-speed = < 115200 >;

	gsm: gsm_mode {
		compatible = "zephyr,gsm-ppp";
	};
};
```

However, I still believe that since the `reel_board` target is used in sample documentation as a `template` target, it should work out of the box. Therefore, an overlay similar to the one shown above should be added to the sample.

jukkar, 2024-05-06:
> Therefore, an overlay similar to the one shown above should be added to the sample.

The gsm-modem sample is deprecated and the cellular-modem sample should be used instead. Most probably no new development is done for gsm-modem although I suppose patches could be accepted (depending on maintainer opinion) if you decide to send a PR fixing this.

nashif, 2024-06-15:
sample was removing, so closing this. https://github.com/zephyrproject-rtos/zephyr/pull/72342

closed 2024-06-15 by nashif