zephyrproject-rtos/zephyr #72344: tests/subsys/logging/log_timestamp/ failing on qemu_x86[_64] on main

aescolar, 2024-05-06:
**Describe the bug**
tests/subsys/logging/log_timestamp/logging.output.custom_timestamp is failing in main since #69705 was merged

**To Reproduce**
Steps to reproduce the behavior:
1. twister -p qemu_x86 -T tests/subsys/logging/log_timestamp/ -s logging.output.custom_timestamp 
4. See error

**Expected behavior**
No errors

**Impact**
main CI broken

**Logs and console output**
https://github.com/zephyrproject-rtos/zephyr/actions/runs/8951164071/job/24587211348#step:11:3389

**Environment (please complete the following information):**
 - OS: Linux Ubuntu 22.04 (CI)
 - Toolchain  Zephyr SDK (CI)
 - Commit SHA or Version used: fa7c05cf4718be15adad0e913fa51a2d5b5bf3dc

**Additional context**
Introduced by https://github.com/zephyrproject-rtos/zephyr/commit/7c03e5de7f0298e9219f82da52b75696c45baa9d

najumon1980, 2024-05-06:
The regression is due to following reason:
sys_clock_hw_cycles_per_sec() used in log_core_init() which invoked during cstart (EARLY init) but HPET driver init is PRE_KERNEL_2. So the z_clock_hw_cycles_per_sec need to init with some default CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC (though it is not right) or ideally need to init z_clock_hw_cycles_per_sec run time during cstart (EARLY init).

najumon1980, 2024-05-06:
Provided following fix for the regression: 
https://github.com/zephyrproject-rtos/zephyr/pull/72352

PR #72345 (2024-05-06): Revert 69705 and fix #72344
https://github.com/zephyrproject-rtos/zephyr/pull/69705
Introduced a regression in main in which tests/subsys/logging/log_timestamp started failing. 

Let's revert the PR. It can be submitted after with the issue fixed.

Fixes #72344
closed 2024-05-06 by jhedberg