zephyrproject-rtos/zephyr #72359: Get function for bt_audio_codec_qos_pref for BAP endpoints

alexsven, 2024-05-06:
The application should have a way of knowing the preferred QoS settings from unicast_servers without using the internal header files. A get function similar to, or included in, the `bt_bap_ep_get_info()` would be nice.

Thalley, 2024-05-06:
@alexsven please follow the Enhancement issue template (https://github.com/zephyrproject-rtos/zephyr/issues/new?assignees=&labels=Enhancement&projects=&template=002_enhancement.md&title=) when creating enhancement requests

alexsven, 2024-05-06:
I thought I did follow it, but I guess it wasn't clear and concise enough? Or should I not have removed the headlines?

Thalley, 2024-05-06:
> I thought I did follow it, but I guess it wasn't clear and concise enough? Or should I not have removed the headlines?

Ideally the headlines would have been kept, so that it's easier to what is the answer to each question/statement :)

closed 2024-05-24 by nashif