zephyrproject-rtos/zephyr #72363: The postfix for the hostname is not configured to the desired value

VineetaNarkhede-eaton, 2024-05-06:
**Describe the bug**
I have enabled hostname configurations and setting postfix, for example "abcd", but not getting this hostname appended, hex value of "abcd" i.e 61626364 is appended to the hostname. Hence the new hostname becomes "zephyr61626364" instead of "zephyrabcd". If I remove the logix of hex conversion then I get desired hostname i.e "zephyrabcd"

Target: STM32U575
Zephyr version: 3.2.0,  Latest version 3.6.0 is also having same code implementation for net_hostname_set_postfix(..)

**One query related to Hostname configuration**:
Can I keep hostname empty like CONFIG_NET_HOSTNAME ="", and set only postfix? What will be the impact?
I tried this and able to use postfix(only) as a hostname.

**To Reproduce**
Steps to reproduce the behavior:
Enable following configurations

CONFIG_NET_HOSTNAME_ENABLE=y
CONFIG_NET_HOSTNAME_UNIQUE=y
CONFIG_NET_HOSTNAME="zephyr"
CONFIG_MDNS_RESPONDER=y
CONFIG_MDNS_RESPONDER_LOG_LEVEL_DBG=y
CONFIG_NET_HOSTNAME_LOG_LEVEL_DBG=y

Set postfix in a code:
const uint8_t* postfix_str = (const uint8_t*)"abcd";
reth = net_hostname_set_postfix(postfix_str, strlen((const char*)postfix_str));

**Expected behavior**
Hostname should be "zephyrabcd"

**Impact**
Impact on the user experience who will be using our product, user won't be able to use the actual postfix set by him/her, so could not progress ahead with hex converted postfix.

**Environment:**
 - OS: Windows
 - Toolchain Zephyr SDK zephyr-sdk-0.15.2

jukkar, 2024-05-07:
Originally the postfix string was suppose to be a MAC address or similar binary data, so a hex value was generated from it. I suppose we could have a normal string used as a postfix too, it would need to be a separate function that instead of using a hex value would use the string directly. As this API is a designed to work this way, I am removing the Bug and replacing it with Enhancement. Would you be able to send such a PR that creates a function to support your use case?

VineetaNarkhede-eaton, 2024-05-08:
Thanks @jukkar!
I will plan to implement a function and send a PR.

Would you mind addressing my other question? (shared in description and also quoted below)
> One query related to Hostname configuration:
> Can I keep hostname empty like CONFIG_NET_HOSTNAME ="", and set only postfix? What will be the impact?
> I tried this and able to use postfix(only) as a hostname.

jukkar, 2024-05-08:
I am afraid the `CONFIG_NET_HOSTNAME` is expected to have some value if the hostname support is enabled.

VineetaNarkhede-eaton, 2024-05-08:
Could you please confirm if there will be any impact if we keep the CONFIG_NET_HOSTNAME empty.

jukkar, 2024-05-08:
If you search `CONFIG_NET_HOSTNAME` in the codebase, it is used for example in DHCPv4 and in DNS (mDNS / LLMNR), both expect it to contain a proper value (len > 0).

VineetaNarkhede-eaton, 2024-05-09:
@jukkar  I am seeking advice on where to incorporate the logic for appending the postfix string without any conversion. what would be better approach:

**Approach 1:**
Modify existing function net_hostname_set_postfix(), create a CONFIG to enable or disable hex conversion

net_hostname_set_postfix(postfix_string)
{
    #if defined(CONFIG_HOSTNAME_HEX_CONV)
     //Convert the postfix_string into hex and append to hostname
    #else
     //Append postfix string to hostname as is
}

**Approach 2:**
Create new function net_hostname_set_postfix_new(postfix_string) for appending postfix string to hostname as is.

**Also, I want to do the special character validation of string, whether validation should happen inside Postfix function or before calling postfix, what is your recommendation.**

jukkar, 2024-05-29:
> **Approach 2:**
> Create new function net_hostname_set_postfix_new(postfix_string) for appending postfix string to hostname as is.

This is the best option so that we can leave the existing API as is. The `_new` suffix is not good for an API name but that can be tweaked in the PR.

VineetaNarkhede-eaton, 2024-06-27:
Hi @jukkar,

Following is the function that we have developed and tested to fix this issue, I will be sending a PR with this function. It appends the postfix string to the hostname without conversion. The implementation is identical to net_hostname_set_postfix() except image size check and hex conversion.

```
int net_hostname_set_postfix_str(const uint8_t *hostname_postfix,
			     int postfix_len)
{
#if !defined(CONFIG_NET_HOSTNAME_UNIQUE_UPDATE)
	static bool postfix_set;
#endif
	int net_hostname_len = sizeof(CONFIG_NET_HOSTNAME) - 1;
	int hostname_len_remain = (NET_HOSTNAME_MAX_LEN - net_hostname_len) + 1;

#if !defined(CONFIG_NET_HOSTNAME_UNIQUE_UPDATE)
	if (postfix_set) {
		return -EALREADY;
	}
#endif

	NET_ASSERT(postfix_len > 0);

	if (hostname_len_remain < postfix_len) {
		NET_DBG("Hostname postfix length %d is exceeding limit of %d", postfix_len, hostname_len_remain);
		return -EMSGSIZE;
	}

	snprintk(&hostname[net_hostname_len], hostname_len_remain, "%s", hostname_postfix);

	NET_DBG("New Unique hostname: %s", hostname);

#if !defined(CONFIG_NET_HOSTNAME_UNIQUE_UPDATE)
	postfix_set = true;
#endif

	return 0;
}
```

PR #75168 (2024-07-28): net: hostname: Add a function to set the hostname postfix as is
Fixes: #72363

The existing function to set the postfix is converting postfix string to hexadecimal. Adding a new function to handle a use case where the provided postfix should be used as is without any conversion.
commit 232c802c (2024-07-28): net: hostname: Add a function to set the hostname postfix as is.

Fixes: #72363

The existing function to set the postfix is converting postfix
string to hexadecimal. Adding a new function to handle a use case
where the provided postfix should be used as is without any conversion.
closed 2024-07-28 by nashif