zephyrproject-rtos/zephyr #72365: drivers: sdhc: Not checking CMD12 return

hellesvik-nordic, 2024-05-06:
**Describe the bug**
If sdhc_spi_request() tries to send a cmd and fails, it may do [CMD12](https://github.com/nrfconnect/sdk-zephyr/blob/b5cf3040298410845311ff7397374540d3451528/drivers/sdhc/sdhc_spi.c#L630-L637). However, the return from this second send command is not handled. If that second CMD12 also fails, it will not be caught by the driver.

**Expected behavior**
The CMD12 should have error handling.

**Context**
I work with Nordic Semiconductor DevZone, and a user asked me about this over there. Private ticket so I can not link it here.
The user I have reports that they see the send failing twice.

danieldegrasse, 2024-05-14:
> The CMD12 should have error handling.

For the case where there was already an error and CMD12 is being sent to abort an in progress transfer, this is expected behavior. I do not want a case where:
- transfer fails
- CMD12 is sent to abort the transfer, and succeeds
- return code to stack now indicates that the transfer as a whole succeeded

Could you clarify what the specific issue the customer is running into as a result of this behavior is? I could perhaps see a case where missing an error on CMD12 when the command being sent is `SD_READ_MULTIPLE_BLOCK` causes an issue, but in the case where the command itself fails, an error on CMD12 shouldn't really change the outcome of the API call should it?

clamattia, 2024-05-14:
> Could you clarify what the specific issue the customer is running into as a result of this behavior is? I could perhaps see a case where missing an error on CMD12 when the command being sent is [...]

I can maybe give my point of view. This happens in a loop with retries. The comment suggests, that `CMD12 is required after multiple read, or to retry failed transfer`, but what happens if the CMD12 itself fails? Currently the retries happen. I would expect to either fail the entire call or to retry the CM12 specifically. I do not know about the specifics of CMD12 though. So take with a grain of salt, maybe my assumptions are wrong.

Behavior:
- `sdhc_spi_send_cmd`
- `CMD12` fails
- `sdhc_spi_send_cmd` is retried.

Ecpected:
- `sdhc_spi_send_cmd`
- `CMD12` fails
- `sdhc_spi_send_cmd` is NOT retried (Maybe either retry `CMD12` or fail everything)

Thanks and cheers.

danieldegrasse, 2024-05-14:
> I can maybe give my point of view. This happens in a loop with retries. The comment suggests, that CMD12 is required after multiple read, or to retry failed transfer, but what happens if the CMD12 [...]

Thank you for the context here- I don't have access to any SD cards that fail CMD12, so the perspective here is valuable. CMD12 can kind of be understood as an "error correction" command- it moves the card back into the transfer state (a known good state)

When performing multiple block reads, CMD12 should be sent to stop the active transfer, and then we must wait for the card to clear the busy signal, so CMD12 could fail. If CMD12 fails in this case, propagating an error code to the stack makes sense.

For the case where CMD12 is being sent after an error (the SD spec explicitly requires this if a Write Multiple Block command fails), then CMD12 failing should (IMO) not really matter. In this case, we already have an error code from the Write Multiple Block command, which is more likely to reflect the root cause of the failure.

To summarize, do you think the following behavior makes sense?

Standard case:
- `sdhc_spi_send_cmd` or `sdhc_spi_write/read_data` fails
- `CMD12` is sent, return code is not checked
- return code from `sdhc_spi_send_cmd` or `sdhc_spi_write/read_data` is returned

Read Multiple Blocks:
- `sdhc_spi_send_cmd` and `sdhc_spi_read_data` succeed
- `CMD12` is sent, return code is checked
- return code from `CMD12` is returned

clamattia, 2024-05-14:
> Standard case:
> 
>     * `sdhc_spi_send_cmd` or `sdhc_spi_write/read_data` fails
> 
>     * `CMD12` is sent, return code is not checked
> 
>     * return code from `sdhc_spi_send_cmd` or `sdhc_spi_write/read_data` is returned

I almost agree. The problem is, that this is not the current behavior as far as I can tell the function performs retries. That means, that the current behavior for single block is:
- `sdhc_spi_send_cmd` ... fails
- `CMD12` may fail
- retries happen nonetheless

I would expect retries to happen only if `CMD12` succeeds or `CMD12` to be retried first.

clamattia, 2024-05-14:
I agree with the rest

danieldegrasse, 2024-05-14:
> I almost agree. The problem is, that this is not the current behavior as far as I can tell the function performs retries. That means, that the current behavior for single block is:
>
>     * [...]

Indeed, it isn't the current behavior- I should have been more clear, sorry. I was suggesting we change the behavior to what I outlined in my comment. Based on your response, it seems like you are ok with that change so I'll go ahead and create a pull request

PR #72754 (2024-05-17): drivers: sdhc: sdhc_spi: release bus on error, and fix CMD12 failue logic
Properly release SPI bus on transmit error within the SDHC SPI driver. In these cases return code is not checked, as we wish to return the error code from the failed transfer to the SD stack.

Fixes #72364

drivers: sdhc: sdhc_spi: rework CMD12 failure logic

Rework CMD12 failure logic for SDHC SPI driver. Previously, the error
code of CMD12 was not checked, so even if CMD12 failed to send the
initial command would be retried. Change this behavior to retry CMD12
until it succeeds. If CMD12 fails, its error code will be p [... cut]
commit a2087bed (2024-05-17): drivers: sdhc: sdhc_spi: rework CMD12 failure logic

Rework CMD12 failure logic for SDHC SPI driver. Previously, the error
code of CMD12 was not checked, so even if CMD12 failed to send the
initial command would be retried. Change this behavior to retry CMD12
until it succeeds. If CMD12 fails, its error code will be propagated to
the caller. Otherwise, the return code from the command being sent by
the caller will be propagated.

Fixes #72365
closed 2024-05-17 by carlescufi