zephyrproject-rtos/zephyr #72391: Integrate ADI HAL module

MaureenHelm, 2024-05-06:
## Origin

Maxim Microcontrollers SDK (MSDK)
https://github.com/analogdevicesinc/msdk
https://github.com/analogdevicesinc/hal_adi

## Purpose

Reuse an existing hardware abstraction layer from the Maxim Microcontrollers SDK to enable Analog Devices SoCs in the MAX32 family. Zephyr shim drivers for all the usual device driver APIs (UART, I2C, SPI, etc.) will be built on top of CMSIS device headers and bare metal drivers provided by the HAL.

## Mode of integration

Integrate as a module, `hal_adi`, following the convention of other SoC vendor HALs.

## Maintainership

@MaureenHelm
@ozersa

## Pull Request

#72291

## Description

The [Maxim Microcontrollers SDK (MSDK)](https://github.com/analogdevicesinc/msdk), now a part of [Analog Devices](https://www.analog.com/en/index.html), contains the necessary software and tools to develop firmware for the [MAX32xxx and MAX78xxx Microcontrollers](https://www.analog.com/en/parametricsearch/10984). This includes register files, peripheral drivers, system startup files, documentation, various utilities, third-party libraries, IDE support files, and a toolchain.

The [ADI HAL](https://github.com/analogdevicesinc/hal_adi) is a pared-down version of the MSDK that contains only the components needed for Zephyr; namely, register files and bare metal peripheral drivers. Utilities, toolchains, IDE support files, and other components not needed by Zephyr are excluded. The ADI HAL is an artifact generated by a GitHub Actions workflow implemented in the MSDK repository.

## Dependencies

CMSIS-Core

## Revision

674d10b0a1aad4cf4b7b90ca52dfd4a494ab9e52

## License

``Apache-2.0``

MaureenHelm, 2024-05-15:
@stephanosio The TSC voted to approve this today ([meeting minutes](https://docs.google.com/document/d/10V00WHKIJIL6DJrKDTimIIeHotD24t0nqf4ESg6fOO4/edit#heading=h.w9k4p9yebfuc)). Can you help to create the module repo in https://github.com/zephyrproject-rtos please?

stephanosio, 2024-05-20:
https://github.com/zephyrproject-rtos/hal_adi has been created.