zephyrproject-rtos/zephyr #72394: LittleFS max file name length is incorrect

dkalowsk, 2024-05-07:
**Describe the bug**

It appears the Zephyr subsys/fs settings for  `FILE_SYSTEM_MAX_FILE_NAME` when combined with LittleFS are not correct.

In the file include/zephyr/fs/fs_interface.h (https://github.com/zephyrproject-rtos/zephyr/blob/3d76e498e9475b67da824247d1fce665f84d0dbc/include/zephyr/fs/fs_interface.h#L37) we can find the following lines:

```
#if !defined(MAX_FILE_NAME) && defined(CONFIG_FILE_SYSTEM_LITTLEFS)
#define MAX_FILE_NAME 256
#endif
```

When looking at the LittleFS headers the following is sethttps://github.com/zephyrproject-rtos/littlefs/blob/408c16a909dd6cf128874a76f21c793798c9e423/lfs.h#L51 

```
// Maximum name size in bytes, may be redefined to reduce the size of the
// info struct. Limited to <= 1022. Stored in superblock and must be
// respected by other littlefs drivers.
#ifndef LFS_NAME_MAX
#define LFS_NAME_MAX 255
#endif
```

As far as I can tell, the Zephyr build system does not set or override the `LFS_NAME_MAX` which results in an off by one byte error for any filesystem name that uses the maximum length possible.

The simple solution here is to fix https://github.com/zephyrproject-rtos/zephyr/include/zephyr/fs/fs_interface.h#L37 to be 255.

While that fixes the size mismatch on a default build, it does not really fix the disconnect that `FILE_SYSTEM_MAX_FILE_NAME` has with the value being used by LittleFS.  Thus it's possible to set a `FILE_SYSTEM_MAX_FILE_NAME` larger than `LFS_NAME_MAX` re-creating the issue, or waste a binary space when a user thinks they have shrunk the MAX_FILE_NAME value.

**To Reproduce**
Steps to reproduce the behavior:

Issue can be reproduced with the `samples/subsys/fs/littlefs` by changing `MAX_PATH_LEN` macro to  `#define MAX_PATH_LEN CONFIG_FILE_SYSTEM_MAX_FILE_NAME` and change `fname1` to something that hits the limit.  

As far as I can tell this issue can be reproduced as far back as 2.7 LTS.

**Expected behavior**
Not to overwrite the bounds of the allocated array.

**Impact**

Provided a pathname is less than 255 no problem.  Once a pathname hits 256 memory starts to go... funny which causes a crash.

nashif, 2024-06-04:
@de-nordic  can you submit a PR? seems like you have it in a branch somewhere

de-nordic, 2024-06-04:
> @de-nordic can you submit a PR? seems like you have it in a branch somewhere

Sorry failed to link PR to issue.

PR #73596 (2024-06-04): Fix for issue  #72394 and additional checks for CONFIG_FILE_SYSTEM_MAX_FILE_NAME
Two commits: 
  1) Fix for MAX_FILE_NAME defaulting for LittleFS to 256, while max allowed is now 255
  2) improved checks of CONFIG_FILE_SYSTEM_MAX_FILE_NAME

Fixes #72394
commit 02dfda81 (2024-06-04): fs: Fix MAX_FILE_NAME setting for LittleFS

Fix MAX_FILE_NAME for LittleFS, which was set to 255, while currently
allowed is 255.

Fixes #72394
closed 2024-06-04 by nashif