zephyrproject-rtos/zephyr #72411: nanopb: build fails when used in external module

sageve, 2024-05-07:
**Describe the bug**

The documentation describes how to use nanopb:

https://docs.zephyrproject.org/latest/services/serialization/nanopb.html

This works fine as long as it is used in the main CMakeLists.txt. If the same procedure is used in an external module the build fails with 

```

CMake Error at C:/Users/KleiserS/Desktop/test/example/deps/modules/lib/nanopb/extra/FindNanopb.cmake:407 (target_compile_features):
  target_compile_features no known features for C compiler

  "GNU"

  version 12.2.0.

```

This happend in general before in #70516, was reverted and fixed later in #70658. I am not sure if it is fixed completely, because my module depending on nanopb fails now.

Commit 6b27cb2b835cd817c191ae9fe1471065c2693d1a is the last one working.

**To Reproduce**
I have prepared an example project. Steps to reproduce this behavior:
1. git clone https://github.com/sageve/nanopb-example
2. cd nanopb-example
3. west init -l
4. west update
5. west build -b qemu_cortex_a9 or any other board

**Expected behavior**
The build should not fail. In case I am misusing nanopb, it should be documented how to use it in modules.

pdgendt, 2024-05-07:
The issue appears to be that the C standard isn't available yet when loading modules.

Moving the following to the end of `nanopb-example/CMakeLists.txt`, works fine.

```cmake
list(APPEND CMAKE_MODULE_PATH ${ZEPHYR_BASE}/modules/nanopb)
include(nanopb)
```

CC @tejlmand

pdgendt, 2024-05-08:
@sageve PTAL at the linked PR, it should fix your issue

PR #72423 (2024-05-13): cmake: Set C compile features early for modules
If modules require C compiler features, these need to be set before calling `add_subdirectory`.

Fixes #72411
closed 2024-05-13 by carlescufi