zephyrproject-rtos/zephyr #72425: Bluetooth: Controller: Missing dependency for BT_CTLR_CENTRAL_ISO and BT_CTLR_PERIPHERAL_ISO when !BT_LL_SW_SPLIT

Thalley, 2024-05-07:
**Describe the bug**
Both `BT_CTLR_CENTRAL_ISO` and `BT_CTLR_PERIPHERAL_ISO` can be enabled without `BT_CONN` is enabled, which can cause build errors. 

**To Reproduce**
Steps to reproduce the behavior:
1. Without enabling `BT_CONN` (e.g. do not enable `BT_CENTRAL` or `BT_PERIPHERAL`), enable `BT_LL_SW_SPLIT` and `BT_CTLR_CENTRAL_ISO` or `BT_CTLR_PERIPHERAL_ISO`.
2. make
3. See error

**Expected behavior**
Expect the build system to ignore the `BT_CTLR_CENTRAL_ISO=y` or `BT_CTLR_PERIPHERAL_ISO=y` and give a warning that it could not be applied when `BT_LL_SW_SPLIT=y`. 

**Impact**
Annoyance

**Logs and console output**
N/A

**Environment (please complete the following information):**
 - Commit SHA or Version used: 33c253c813d8cc56209ce0b572e127cd6eaa941b

**Additional context**
N/A

cvinayak, 2024-05-07:
We use the follow construct:
```
config BT_CTLR_CENTRAL_ISO
	bool "LE Connected Isochronous Stream Central" if !BT_LL_SW_SPLIT
	depends on BT_CTLR_CENTRAL_ISO_SUPPORT && BT_CENTRAL
	default y if BT_ISO_CENTRAL
	help
	  Enable support for Bluetooth 5.2 LE Connected Isochronous Stream
	  Central role in the Controller.
 
config BT_CTLR_CENTRAL_ISO
	bool "LE Connected Isochronous Stream Central (Split Link Layer) [EXPERIMENTAL]" if BT_LL_SW_SPLIT
	select BT_CTLR_SET_HOST_FEATURE
	select EXPERIMENTAL if BT_LL_SW_SPLIT
 
config BT_CTLR_PERIPHERAL_ISO
	bool "LE Connected Isochronous Stream Peripheral" if !BT_LL_SW_SPLIT
	depends on BT_CTLR_PERIPHERAL_ISO_SUPPORT && BT_PERIPHERAL
	default y if BT_ISO_PERIPHERAL
	help
	  Enable support for Bluetooth 5.2 LE Connected Isochronous Stream
	  Peripheral role in the Controller.
 
config BT_CTLR_PERIPHERAL_ISO
	bool "LE Connected Isochronous Stream Peripheral (Split Link Layer) [EXPERIMENTAL]" if BT_LL_SW_SPLIT
	select BT_CTLR_SET_HOST_FEATURE
	select EXPERIMENTAL if BT_LL_SW_SPLIT
 
```

This leads to the `depends on` being OR-ed, as seen in the `menuconfig` 's help:
```
ame: BT_CTLR_CENTRAL_ISO
Prompt: LE Connected Isochronous Stream Central
Prompt: LE Connected Isochronous Stream Central (Split Link Layer) [EXPERIMENTAL]
Type: bool
Value: y
 
Help:
 
  Enable support for Bluetooth 5.2 LE Connected Isochronous Stream
  Central role in the Controller.
 
Direct dependencies (=y):
     BT_CTLR_CENTRAL_ISO_SUPPORT(=y) && BT_CENTRAL(=y) && BT_CTLR(=y) && BT_HCI(=y) && BT(=y)  (=y)
  || BT_CTLR(=y) && BT_HCI(=y) && BT(=y)  (=y)
 
Default:
  - y
    Condition (=y):
       BT_ISO_CENTRAL(=y)
 ```
 

 Possible solution is to duplicate the `depends on BT_CTLR_PERIPHERAL_ISO_SUPPORT && BT_PERIPHERAL`  for `BT_LL_SW_SPLIT` = y

aescolar, 2024-05-07:
@cvinayak would it be time to remove the experimental definitions?

cvinayak, 2024-05-07:
> @cvinayak would it be time to remove the experimental definitions?

No, upstream implementation is not feature complete nor fully conformance tested for ISO features.

closed 2025-03-24 by kartben