zephyrproject-rtos/zephyr #72459: DNS-SD Service Discovery

besmarsh, 2024-05-08:
**Is your enhancement proposal related to a problem? Please describe.**

DNS-SD (DNS-based Service Discovery) is a standard method for enabling discovery of services on a local IP network. DNS-SD consists of two parts, Service **Advertisement**, and Service **Discovery**. Zephyr currently supports the Service **Advertisement** part of DNS-SD, but does not support the Service **Discovery** part.

DNS-SD is one of the 3 core aspects of Zero-Configuration Networking, along with IP Autoconfiguration and mDNS, both of which are supported by Zephyr. The Service **Discovery** part of DNS-SD is the missing part to enable a Zephyr device to fully work in a Zero-Configuration Networking environment.

**Describe the solution you'd like**

This enhancement proposal is for implementing the Service **Discovery** part of DNS-SD. Much of the groundwork is already present, as DNS-SD builds on top of standard DNS, which is already implemented.

The implementation should likely provide an API to discover services through DNS-SD, similar to the existing DNS API for resolving hostnames. As multiple instances can exist for a given service, a consumer of the API should in some way be provided with a list of discovered service instances, from which a service instance can be chosen and resolved using the standard DNS API.

The DNS-SD implementation should work with both unicast DNS and multicast DNS (mDNS).

**Describe alternatives you've considered**

DNS-SD is the accepted standard solution for local IP network service discovery. I don't know of any alternatives which would be better to support.

**Additional context**

* [RFC 6763](https://datatracker.ietf.org/doc/html/rfc6763) is the IETF RFC for DNS-SD
* Some experimentation was done earlier this year by another Zephyr contributor - https://github.com/Ayush1325/zephyr/tree/zeroconf

Ayush1325, 2024-05-18:
Great proposal. I would love this in the mainline.

I had to pause the previous work since I came across some problems with networking on beagleconnect freedom in Zephyr mainline (https://github.com/zephyrproject-rtos/zephyr/issues/71191, https://github.com/zephyrproject-rtos/zephyr/issues/68674). I can pick it up again once I get all the other parts of networking on bcf resolved. Of course, it is also fine if someone else wants to pick it up (since it might take me a few weeks to get around to working on this).

erikmansson, 2025-02-28:
@Ayush1325 Did you ever get around to this again?

Ayush1325, 2025-02-28:
> [@Ayush1325](https://github.com/Ayush1325) Did you ever get around to this again?

Not really. And I don't think I will come around to working on this anytime soon. There might be a gsoc student working on greybus this summer though. So he might have to tackle this.