zephyrproject-rtos/zephyr #72471: Bluetooth OTS asserts or fail on runtime depending on configuration options

sjanc, 2024-05-08:
**Describe the bug**
Currently OTS code asserts or fails depending on configuration selected in Kconfig.
Eg enabling CONFIG_BT_OTS_OACP_CHECKSUM_SUPPORT results in assert on init for instances that doesn't implement obj_cal_checksum callback.  Similarly bt_ots_obj_add() may fail if added object has unsupported (ie. disabled in Kconfig) feature.

Since Kconfig is global configuration this leads to confusion where multiple OTS instances are created (eg some might be internal like in MCS case) user might not have control over what is implemented/required by OTS instance...

Either optional features should be made mandatory (ie no kconfig option to disable or always require to implement all callbacks) or some runtime disablement (eg masking out disabled features) could be added.   Otherwise one may end up in case where is it impossible to use multiple OTS instances with contradicting features enabled.

note: There are also few other issues with OTS  API (ie no userdata for callbacks) so maybe this issue should cover more overhaul as enhancement instead of bug?

aescolar, 2024-05-14:
CC @sjanc @Thalley @joerchan @asbjornsabo

Thalley, 2024-07-03:
Looking a bit more into this, this is not a bug, but a (poor) design choice. Checksum, like supporting object creation, deletion, etc. is per the spec an optional feature

[image]

It gets a bit more complicated than that, since the only way an OTS instance can have any object is by creating them, and thus the creation and deletion of objects as local operations are mandatory, similar to read and write. 

So while 
```c
	__ASSERT(ots_init->cb->obj_created,
		 "Callback for object creation is not set");
	__ASSERT(ots_init->cb->obj_deleted ||
		 !BT_OTS_OACP_GET_FEAT_CREATE(ots_init->features.oacp),
		 "Callback for object deletion is not set and object creation is enabled");
#if defined(CONFIG_BT_OTS_OACP_CHECKSUM_SUPPORT)
	__ASSERT(ots_init->cb->obj_cal_checksum,
		 "Callback for object calculate checksum is not set");
#endif
	__ASSERT(ots_init->cb->obj_read ||
		 !BT_OTS_OACP_GET_FEAT_READ(ots_init->features.oacp),
		 "Callback for object reading is not set");
	__ASSERT(ots_init->cb->obj_write ||
		 !BT_OTS_OACP_GET_FEAT_WRITE(ots_init->features.oacp),
		 "Callback for object write is not set");
```

Makes sense in terms of the OACP (remote procedures), we probably shouldn't treat Checksum similar to the others, as we treat local and remote procedure the same. 

I'll create a PR to optionally support these callbacks, but for all them so they are continued to be treated similar.

Thalley, 2024-07-03:
Looked further into it, and think I found the issue. The checks for callbacks should be conditional on the `ots_init->features.oacp` which weren't properly done. The callbacks are actually checked to be non-NULL before calling them, so they are allowed to be NULL.

closed 2024-07-06 by aescolar