zephyrproject-rtos/zephyr #72491: doc: Documentation issue in 'boards/nxp/imx8mp_evk/doc/index'

flobro30101, 2024-05-08:
mimx8mp_evk_a53 and mimx8mp_evk_m7 board configurations as described in the document are not present in the sdk-0.16.6

The board configurations are not available in the newest SDK. Do they exist somewhere else?

[https://docs.zephyrproject.org/latest/boards/nxp/imx8mp_evk/doc/index.html](https://docs.zephyrproject.org/latest/boards/nxp/imx8mp_evk/doc/index.html)

flobro30101, 2024-05-08:
I now understand that the name listed in the document page was not explicit enough, and have realized that the board config files must be one of:
imx8mp_evk/mimx8ml8/adsp
imx8mp_evk/mimx8ml8/a53
imx8mp_evk/mimx8ml8/a53/smp
imx8mp_evk/mimx8ml8/m7
imx8mp_evk/mimx8ml8/m7/ddr

so the base board name is imx8mp_evk. So maybe the document can be more explicit (anyway I am a newbie with this build system, so it was not clear to me). 

I do see the Note*
"A board may contain one or multiple SoCs, Also, each SoC may contain one or more CPU clusters. "... in the "Getting started guide" which helped when I started exploring the "board" subdirectory and saw the different imx8mp entries.

So this is more of an enhancement request rather than a bug. Sorry!

nordicjm, 2024-05-09:
The documentation is wrong and need to be updated

flobro30101, 2024-05-09:
I see that the doc was updated already... I thought I was seeing things (or going crazy) until I saw the "last generated  date".

It is now very clear. Very nice. Thank you

closed 2024-05-09 by flobro30101