zephyrproject-rtos/zephyr #72536: RTC Test Additions

RichardSWheatley, 2024-05-09:
Current RTC testing for alarm includes only hour and minute.
Some RTCs do not allow such alarms and require more configs.

This Issue was formed from discussion below where the consensus was to have test coverage for RTC Full Config Alarm and for Partial Config Alarm _(may or not be the ending names for the tests)_.

### Discussed in https://github.com/zephyrproject-rtos/zephyr/discussions/72480

<div type='discussions-op-text'>

<sup>Originally posted by **RichardSWheatley** May  8, 2024</sup>
Hello, @bjarki-trackunit!

You are the maintainer for the RTC section.
I am writing the Ambiq RTC driver when I noticed that the tests for RTC make some assumptions that are not true for the Ambiq RTC.

1. We do not support 99-00 rollover which I already fixed in the tests by adding a check for EINVAL.
```
	if (ret == -EINVAL) {
		TC_PRINT("Rollover not supported\n");
		ztest_test_skip();
	} else {
		zassert_ok(ret, "RTC Set Time Failed");
	}
```
2. When setting the alarms, for the Ambiq RTC you must set the entire struct with a non-zero value

This makes our implementation fail the most of the testing without additions to the tests.
I do not like to make additions to tests unless it is absolutely needed.

Do you think there is a better way to accomplish this?

[image]
[image]

</div>

RichardSWheatley, 2024-05-13:
@bjarki-trackunit I have started the work but have gotten busy at my job.
I am commenting here so that this won't be marked stale.

bjarki-trackunit, 2024-05-13:
Don't worry, the stale bot waits for 60 days :) also, I'm invested in the addition so if you don't have time I will gladly add it :)

RichardSWheatley, 2024-05-22:
> Don't worry, the stale bot waits for 60 days :) also, I'm invested in the addition so if you don't have time I will gladly add it :)

I'm going to be honest, I do not have the time since my company is putting out new software releases monthly.

bjarki-trackunit, 2024-05-22:
Noted, I will find time to do it :)

bjarki-trackunit, 2024-06-05:
When you find the time, I think this PR will be able to solve the alarm time mask issue :) https://github.com/zephyrproject-rtos/zephyr/pull/73789

Simply add
```
CONFIG_TEST_RTC_ALARM_TIME_MASK=79
```
to you .conf file in `boards/` and it should set every field the Ambiq needs

RichardSWheatley, 2024-06-16:
> When you find the time, I think this PR will be able to solve the alarm time mask issue :) https://github.com/zephyrproject-rtos/zephyr/pull/73789
> 
> 
> 
> Simply add
> 
> ```
> 
> CONFIG_TEST_RTC_ALARM_TIME_MASK=79
> 
> ```
> 
> to you .conf file in `boards/` and it should set every field the Ambiq needs

I did not get an email when you left this comment. 
I will test tomorrow. Sorry for the delay and thank you!

RichardSWheatley, 2024-06-18:
> When you find the time, I think this PR will be able to solve the alarm time mask issue :) #73789
> 
> Simply add
> 
> ```
> CONFIG_TEST_RTC_ALARM_TIME_MASK=79
> ```
> 
> to you .conf file in `boards/` and it should set every field the Ambiq needs

START - test_alarm

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/rtc/rtc_api/src/test_alarm.c:120: rtc_api_test_alarm: (ret is non-zero)
Failed to set alarm 0 time
 FAIL - test_alarm in 0.014 seconds
===================================================================
START - test_alarm_callback

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/rtc/rtc_api/src/test_alarm_callback.c:80: rtc_api_test_alarm_callback: (ret is non-zero)
Failed to set alarm 0 time
 FAIL - test_alarm_callback in 0.016 seconds

bjarki-trackunit, 2024-06-18:
Did the test suite work as expected using the mask 119 for the ambiq?

RichardSWheatley, 2024-06-18:
> Did the test suite work as expected using the mask 119 for the Ambiq?

It is giving errors but they appear unrelated to your changes.
Give me until next Monday while I work through some non-Zephyr related code.

bjarki-trackunit, 2024-06-18:
ok, I will until you are certain the issue is fixed, then we can move forward with getting it merged :)

closed 2024-06-29 by nashif