zephyrproject-rtos/zephyr #72546: ESP32/ESP32S3 RMT

sam131208, 2024-05-10:
**Is your feature request related to a problem? Please describe.**
In general, when applying an LED strip, we like to use SPI to transmit the signal to it.Sometimes, however, the transmitted signal is interrupted by other interruptions. In addition, it also occupies additional SPI ports.

**Describe the solution you'd like**
RMT does a better job of driving a small number of LED strips, it is not interrupted by other interrupts, and it does not need to occupy DMA.

**Describe alternatives you've considered**
There is no better way to drive the LED strip than the RMT, when the number of led strips is less than 16, the FIFO using RMT can drive the led strip perfectly.

shenyi97, 2024-10-06:
I just start to work on the RMT driver on ESP32S3. Do you have any advices about the design of RMT api? Is it a good idea to keep it consistant with esp-idf api?

patelvijay12007, 2024-10-15:
Any progress on this driver? We need it to drive neopixel LEDs.

MeisterBob, 2025-01-23:
@shenyi97 are you still working in a RMT driver? Is there anything I can help you with or anything I can use for further development in case you aren't working on this anymore?

joelguittet, 2025-12-19:
@sam131208 @shenyi97 @patelvijay12007 @MeisterBob hello! anyone working on this support? I will appreciate to help to have this support :-)

@sylvioalves maybe you can also help on the status from Espressif? Is support of RMT something you are working on?
I see there are missing parts in hal_espressif (only a a few rmt hal functions in `rmt_hal.c`), and probably a custom driver in zephyr repository? The question of @shenyi97 is interesting: what is the approach you would adopt to add a driver in zephyr, particularly about APIs? Trying to match esp-idf driver APIs or not?

closed 2026-08-21 by kartben