zephyrproject-rtos/zephyr #72556: select GPIO is broken

swift-tk, 2024-05-10:
**Describe the bug**
select GPIO split out dependency loop as of commit id 422092f2d3f.
going back to the last one 80c5f72fe26f44 have no such problem.

This is caused by "select P_DEVICE if PM" in Kconfig.smartbond

**How to reproduce**

Add select GPIO to Kconfig.mcux
```
config MEMC_MCUX_FLEXSPI
	bool
	select PINCTRL
	select GPIO #add
```

Try
west build -p always -b mimxrt595_evk/mimxrt595s/cm33 samples/drivers/memc

 log attached
[log.txt](https://github.com/zephyrproject-rtos/zephyr/files/15273036/log.txt)

swift-tk, 2024-05-10:
@RichardSWheatley @AlessandroLuo for awareness

henrikbrixandersen, 2024-05-10:
@ceolin This is power-management related, caused by `select PM_DEVICE if PM` in `drivers/gpio/Kconfig.smartbond`.

henrikbrixandersen, 2024-05-10:
Introduced in https://github.com/zephyrproject-rtos/zephyr/pull/61857

henrikbrixandersen, 2024-05-10:
This can be reproduced on `main` without introducing changes by running:
`west build -b ucans32k1sic -p always tests/drivers/can/api/`

carlescufi, 2024-05-10:
@sjanc this was introduced by Smartbond.

sjanc, 2024-05-10:
@kasjer FYI

swift-tk, 2024-05-13:
@andrzej-kaczmarek Hi, a quick fix is appreciated as my PR kind of depending on select GPIO to compile.

kasjer, 2024-05-13:
@swift-tk working on it, stay tuned...

PR #72663 (2024-05-14): soc: smartbond: Move PM_DEVICE dependency to soc
For DA1469x if PM config is selected PM_DEVICE must also be selected for GPIO to work when device enters/exists deep sleep.

Previously GPIO and regulator drivers selected PM_DEVICE when PM was enabled.
Now it is moved to SOC instead.

PM_DEVICE selection in GPIO could result in circular dependency for mcux if MEMC_MCUX_FLEXSPI (which is already dependent on PM_DEVICE) was to be additionally dependent on GPIO.

This should fix #72556
closed 2024-05-14 by aescolar