zephyrproject-rtos/zephyr #72572: tests/drivers/gpio/gpio_basic_api fails for imx93_evk/mimx9352/a55 target

nordic-krch, 2024-05-10:
**Describe the bug**
`tests/drivers/gpio/gpio_basic_api/drivers.gpio.2pin` fails to compile for `imx93_evk/mimx9352/a55`

**To Reproduce**
Steps to reproduce the behavior:
1. `west build -p -b imx93_evk/mimx9352/a55 tests/drivers/gpio/gpio_basic_api/ -T drivers.gpio.2pin`
2. See error

**Expected behavior**
Test shall compile or configuration be removed.

**Impact**
CI failures.

**Environment (please complete the following information):**
 - Commit 2e072ad017cf41673e7f5a7ff44aebb4624e4ee7

aescolar, 2024-05-14:
@nordic-krch can you please link to the PR CI in which this fails?

ubieda, 2024-05-14:
@aescolar I was able to replicate it locally on `nordic-krch:nrf54h_gpio`, but it's not happening on `upstream/main`. Can you try rebasing and close if it's no longer an issue, @nordic-krch?

dleach02, 2024-05-14:
Fixed with https://github.com/zephyrproject-rtos/zephyr/pull/72644

closed 2024-05-14 by dleach02