zephyrproject-rtos/zephyr #72588: Blobs: Introduce NXP binary blobs for IW612 and RW612

mmahadevan108, 2024-05-10:
## Origin

The following binary blobs are to be introduced:

```
1. uart_nw61x_se.h
2. rw61x_sb_ble_15d4_combo_a2.bin
3. rw61x_sb_ble_a2.bin
4. rw61x/rw61x_sb_wifi_a2.bin
```

## Type

- [ ] Precompiled library
- [X] Firmware image

## Module

hal_nxp: [https://github.com/zephyrproject-rtos/hal_nxp/.](https://github.com/zephyrproject-rtos/hal_nxp)

## Purpose

Precompiled firmware (binary blobs) supporting NXP IW612 and RW612 SoC's.

## Pull Request

Zephyr: 
[https://github.com/zephyrproject-rtos/zephyr/pull/70815](https://github.com/zephyrproject-rtos/zephyr/pull/70815)

HAL: 
[https://github.com/zephyrproject-rtos/hal_nxp/pull/381](https://github.com/zephyrproject-rtos/hal_nxp/pull/381)

## Dependencies

None

## License

uart_nw61x_se.h : BSD 3 License
rw61x_sb_ble_15d4_combo_a2.bin, rw61x_sb_ble_a2.bin, rw61x/rw61x_sb_wifi_a2.bin: LA_OPT_NXP_Software_License.txt

dleach02, 2024-05-22:
https://docs.google.com/document/d/10V00WHKIJIL6DJrKDTimIIeHotD24t0nqf4ESg6fOO4/edit
05/22/2024 TSC Meeting:

> David Leach makes a motion to approve these binary blobs. Ian Fyall seconds
> No objection - motion passes.
>

dleach02, 2024-05-31:
binary blobs merged

closed 2024-05-31 by dleach02