zephyrproject-rtos/zephyr #72627: Would it be possible to have support for the XNUCLEO F411RE?

gabohc, 2024-05-12:
The XNUCLEO F411RE is an enhanced version from the NUCLEO F411RE from Waveshare: https://www.waveshare.com/wiki/XNUCLEO-F411RE

**Is your feature request related to a problem? Please describe.**

The XNUCLEO F411RE  is in concept fully compatible out of the box with the NUCLEO F411RE  (same pinouts, periphereals, etc).  The samples and demos from other platforms, such as the Blinky from MBED IoS, perfectly work without issues and without any changes. In Zephyr however, they do not. Related discussion: https://github.com/zephyrproject-rtos/zephyr/discussions/53460

**Describe the solution you'd like**

Support for the XNUCLEO F411RE

**Additional context**

I worked with custom board that basically only shared the MCU from another supported board  96b Aerocore 2) and managed to get it working without nothing more than some minor changes in the DTS as usual, but still built with `-b 96b_aerocore2` directly 

I would be happy to contribute, although I am quite junior! Willing to help, provided the guidance of someone else more experienced

s163555, 2024-05-12:
I just want to throw my support for this issue as to my knowledge from last year I had several key issues working with this board and zephyr. From getting this board flashed to successfully getting an LED blink to run I never managed to resolve it.

erwango, 2024-05-13:
Sure, don't hesitate to contribute the porting of this board.

erwango, 2024-06-27:
Removing myself as assignee as this board is not a ST board.