zephyrproject-rtos/zephyr #72639: ESP32S3 zephyr build example fails on OSX

dcmcshan, 2024-05-12:
I am following the directions on: 
https://docs.zephyrproject.org/latest/boards/espressif/esp32s3_devkitc/doc/index.html

Everything goes smoothly, but the build fails.  Guessing something isn't sized correctly?

`% west build -b esp32s3_devkitc/esp32s3/procpu  samples/hello_world
[0/6] Performing build step for 'hello_world'
ninja: no work to do.
[1/6] Performing build step for 'mcuboot'
[1/6] Linking C executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /Users/danielmcshan/zephyrproject/zephyr/build/mcuboot/zephyr/zephyr_pre0.map 
: && ccache /Users/danielmcshan/zephyr-sdk-0.16.5-1/xtensa-espressif_esp32s3_zephyr-elf/bin/xtensa-espressif_esp32s3_zephyr-elf-gcc  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/xtensa/core/offsets/offsets.c.obj  -fuse-ld=bfd  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/Users/danielmcshan/zephyrproject/zephyr/build/mcuboot/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/xtensa/core/libarch__xtensa__core.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/flash/libdrivers__flash.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  modules/mcuboot/libmcuboot_util.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/Users/danielmcshan/zephyr-sdk-0.16.5-1/xtensa-espressif_esp32s3_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32s3_zephyr-elf/12.2.0"  -L/Users/danielmcshan/zephyrproject/zephyr/build/mcuboot/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -lgcc  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie  -T/Users/danielmcshan/zephyrproject/modules/hal/espressif/zephyr/esp32s3/src/linker/esp32s3.rom.alias.ld  -T/Users/danielmcshan/zephyrproject/modules/hal/espressif/zephyr/esp32s3/../../components/esp_rom/esp32s3/ld/esp32s3.rom.ld  -T/Users/danielmcshan/zephyrproject/modules/hal/espressif/zephyr/esp32s3/../../components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld  -T/Users/danielmcshan/zephyrproject/modules/hal/espressif/zephyr/esp32s3/../../components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld  -T/Users/danielmcshan/zephyrproject/modules/hal/espressif/zephyr/esp32s3/../../components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld  -T/Users/danielmcshan/zephyrproject/modules/hal/espressif/zephyr/esp32s3/../../components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld  -T/Users/danielmcshan/zephyrproject/modules/hal/espressif/zephyr/esp32s3/../../components/soc/esp32s3/ld/esp32s3.peripherals.ld  -Wl,--wrap=longjmp && cd /Users/danielmcshan/zephyrproject/zephyr/build/mcuboot/zephyr && /opt/homebrew/Cellar/cmake/3.29.3/bin/cmake -E true
/Users/danielmcshan/zephyr-sdk-0.16.5-1/xtensa-espressif_esp32s3_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32s3_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32s3_zephyr-elf/bin/ld.bfd: zephyr/zephyr_pre0.elf section `.iram0.text' will not fit in region `iram_seg'
/Users/danielmcshan/zephyr-sdk-0.16.5-1/xtensa-espressif_esp32s3_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32s3_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32s3_zephyr-elf/bin/ld.bfd: section .iram0.loader_text LMA [00000000403bec00,00000000403c048f] overlaps section .iram0.text LMA [00000000403b6000,00000000403c006f]
/Users/danielmcshan/zephyr-sdk-0.16.5-1/xtensa-espressif_esp32s3_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32s3_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32s3_zephyr-elf/bin/ld.bfd: region `iram_seg' overflowed by 5232 bytes
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
[3/6] Completed 'hello_world'
FAILED: images/bootloader/mcuboot-prefix/src/mcuboot-stamp/mcuboot-build /Users/danielmcshan/zephyrproject/zephyr/build/images/bootloader/mcuboot-prefix/src/mcuboot-stamp/mcuboot-build 
cd /Users/danielmcshan/zephyrproject/zephyr/build/mcuboot && /opt/homebrew/Cellar/cmake/3.29.3/bin/cmake --build .
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /opt/homebrew/bin/cmake --build /Users/danielmcshan/zephyrproject/zephyr/build
`

sylvioalves, 2024-05-13:
@dcmcshan Hi, can you provide which revision you have been working with?

marekmatej, 2024-06-11:
Hi @dcmcshan can you please update and try? The S3 issue was addressed in the recently merged  https://github.com/zephyrproject-rtos/zephyr/pull/73480

sylvioalves, 2024-07-10:
Fixed in https://github.com/zephyrproject-rtos/zephyr/pull/73480