zephyrproject-rtos/zephyr #72650: samples: drivers: jesd216: samples/drivers/jesd216/sample.drivers.jesd216 no output for nrf54h20

nordic-piks, 2024-05-13:
**Describe the bug**
The samples/drivers/jesd216/sample.drivers.jesd216 fails - no output from console

Observed for
- nrf54h20dk/nrf54h20/cpuapp

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf54h20dk/nrf54h20/cpuapp connected
2. go to your zephyr dir
3. call ./scripts/twister -T samples/drivers/jesd216 -p nrf54h20dk/nrf54h20/cpuapp--device-testing --device-serial /dev/ttyACM0 -v --inline-logs
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
No output.
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.16.5
 - Commit SHA or Version used: v3.6.0-3753-gbe682e22e166

nordic-piks, 2024-05-13:
@e-rk This started to be executed (and failing) from #71989

e-rk, 2024-05-13:
@nordic-piks Have you powered-up the GPIO P6 and the flash chip with the board configurator?

nordic-piks, 2024-05-13:
> @nordic-piks Have you powered-up the GPIO P6 and the flash chip with the board configurator?

Indeed, that needs to be changed - external flash needs to be powered, will do that.

e-rk, 2024-05-13:
The sample also requires a new nrf-regtool version that configures the CTRLSEL registers needed by the peripheral.

nordic-piks, 2024-05-13:
> The sample also requires a new nrf-regtool version that configures the CTRLSEL registers needed by the peripheral.

https://github.com/nrfconnect/nrf-regtool/pull/133

nordic-piks, 2024-07-02:
Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/73302

closed 2024-07-02 by nordic-piks