zephyrproject-rtos/zephyr #72651: Linking ESP32 with anything which contains mbedtls fail

boaretti, 2024-05-13:
bug:
Linking ESP32 with anything which contains mbedtls fail

Using:

Windows 11
Zephyr version: 3.6.99 / SDK 0.16.5
To Reproduce:
Steps to reproduce the behavior:

...net\cloud\mqtt_azure>west build -b esp32 -p

Expected behavior:
Linking Ok.

Impact:
Showstopper

Logs and console output

mqtt_azure>west build -b esp32 -p
-- west build: making build dir D:\DEV\zephyrproject\zephyr\samples\net\cloud\mqtt_azure\build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: D:/DEV/zephyrproject/zephyr/samples/net/cloud/mqtt_azure
-- CMake version: 3.29.3
-- Found Python3: C:/Python311/python.exe (found suitable version "3.11.9", minimum required is "3.8") found components: Interpreter
-- Cache files will be written to: D:/DEV/zephyrproject/zephyr/.cache
-- Zephyr version: 3.6.99 (D:/DEV/zephyrproject/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
CMake Warning at D:/DEV/zephyrproject/zephyr/cmake/modules/boards.cmake:110 (message):
  Deprecated BOARD=esp32 specified, board automatically changed to:
  esp32_devkitc_wrover/esp32/procpu.
Call Stack (most recent call first):
  D:/DEV/zephyrproject/zephyr/cmake/modules/zephyr_default.cmake:132 (include)
  D:/DEV/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  D:/DEV/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Board: esp32_devkitc_wrover, qualifiers: esp32/procpu
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.5 (D:/DEV/zephyrproject/zephyr-sdk-0.16.5-1)
-- Found toolchain: zephyr 0.16.5 (D:/DEV/zephyrproject/zephyr-sdk-0.16.5-1)
-- Found Dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: D:/DEV/zephyrproject/zephyr/boards/espressif/esp32_devkitc_wrover/esp32_devkitc_wrover_procpu.dts
-- Found devicetree overlay: D:/DEV/zephyrproject/zephyr/samples/net/cloud/mqtt_azure/socs/esp32_procpu.overlay
-- Generated zephyr.dts: D:/DEV/zephyrproject/zephyr/samples/net/cloud/mqtt_azure/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: D:/DEV/zephyrproject/zephyr/samples/net/cloud/mqtt_azure/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: D:/DEV/zephyrproject/zephyr/samples/net/cloud/mqtt_azure/build/zephyr/dts.cmake
Parsing D:/DEV/zephyrproject/zephyr/samples/net/cloud/mqtt_azure/Kconfig
Loaded configuration 'D:/DEV/zephyrproject/zephyr/boards/espressif/esp32_devkitc_wrover/esp32_devkitc_wrover_procpu_defconfig'
Merged configuration 'D:/DEV/zephyrproject/zephyr/samples/net/cloud/mqtt_azure/prj.conf'
Merged configuration 'D:/DEV/zephyrproject/zephyr/samples/net/cloud/mqtt_azure/socs/esp32_procpu.conf'
Configuration saved to 'D:/DEV/zephyrproject/zephyr/samples/net/cloud/mqtt_azure/build/zephyr/.config'
Kconfig header saved to 'D:/DEV/zephyrproject/zephyr/samples/net/cloud/mqtt_azure/build/zephyr/include/generated/autoconf.h'
-- Found GnuLd: d:/dev/zephyrproject/zephyr-sdk-0.16.5-1/xtensa-espressif_esp32_zephyr-elf/xtensa-espressif_esp32_zephyr-elf/bin/ld.bfd.exe (found version "2.38")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: D:/DEV/zephyrproject/zephyr-sdk-0.16.5-1/xtensa-espressif_esp32_zephyr-elf/bin/xtensa-espressif_esp32_zephyr-elf-gcc.exe
esptool path: D:/DEV/zephyrproject/modules/hal/espressif/tools/esptool_py/esptool.py
-- Configuring done (3.7s)
-- Generating done (0.2s)
-- Build files have been written to: D:/DEV/zephyrproject/zephyr/samples/net/cloud/mqtt_azure/build
←[92m-- west build: building application
[3/542] Generating include/generated/version.h
-- Zephyr version: 3.6.99 (D:/DEV/zephyrproject/zephyr), build: v3.6.0-3678-gd1687a557cdc
[537/542] Linking C executable zephyr\zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map D:/DEV/zephyrproject/zephyr/samples/net/cloud/mqtt_azure/build/zephyr/zephyr_pre0.map
C:\Windows\system32\cmd.exe /C "cd . && D:\DEV\zephyrproject\zephyr-sdk-0.16.5-1\xtensa-espressif_esp32_zephyr-elf\bin\xtensa-espressif_esp32_zephyr-elf-gcc.exe  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr\zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/xtensa/core/offsets/offsets.c.obj  -fuse-ld=bfd  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=D:/DEV/zephyrproject/zephyr/samples/net/cloud/mqtt_azure/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/xtensa/core/libarch__xtensa__core.a  zephyr/lib/libc/picolibc/liblib__libc__picolibc.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/lib/posix/options/liblib__posix__options.a  zephyr/subsys/random/libsubsys__random.a  zephyr/subsys/net/libsubsys__net.a  zephyr/subsys/net/l2/ethernet/libsubsys__net__l2__ethernet.a  zephyr/subsys/net/l2/wifi/libsubsys__net__l2__wifi.a  zephyr/subsys/net/ip/libsubsys__net__ip.a  zephyr/subsys/net/lib/mqtt/libsubsys__net__lib__mqtt.a  zephyr/subsys/net/lib/config/libsubsys__net__lib__config.a  zephyr/subsys/net/lib/dhcpv4/libsubsys__net__lib__dhcpv4.a  zephyr/subsys/net/lib/dns/libsubsys__net__lib__dns.a  zephyr/subsys/net/conn_mgr/libsubsys__net__conn_mgr.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/entropy/libdrivers__entropy.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  zephyr/drivers/wifi/libdrivers__wifi.a  modules/mbedtls/libmbedTLSBase.a  modules/mbedtls/libmbedTLSCrypto.a  modules/mbedtls/libmbedTLSX509.a  modules/mbedtls/libmodules__mbedtls.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"d:/dev/zephyrproject/zephyr-sdk-0.16.5-1/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/12.2.0"  -LD:/DEV/zephyrproject/zephyr/samples/net/cloud/mqtt_azure/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  modules/mbedtls/libmbedTLSBase.a  modules/mbedtls/libmbedTLSCrypto.a  modules/mbedtls/libmbedTLSX509.a  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie  -DPICOLIBC_DOUBLE_PRINTF_SCANF  -TD:/DEV/zephyrproject/modules/hal/espressif/zephyr/esp32/src/linker/esp32.rom.alias.ld  -TD:/DEV/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.ld  -TD:/DEV/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.api.ld  -TD:/DEV/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.libgcc.ld  -TD:/DEV/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld  -TD:/DEV/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld  -TD:/DEV/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/soc/esp32/ld/esp32.peripherals.ld  -Wl,--wrap=longjmp  -lnet80211  -lcore  -lpp  -lphy  -lrtc  -lgcc  -LD:/DEV/zephyrproject/modules/hal/espressif/zephyr/esp32/../blobs/lib/esp32  --specs=picolibc.specs  -lc  -lgcc && C:\Windows\system32\cmd.exe /C "cd /D D:\DEV\zephyrproject\zephyr\samples\net\cloud\mqtt_azure\build\zephyr && "C:\Program Files\CMake\bin\cmake.exe" -E true""
[... 6 lines cut ...]
d:/dev/zephyrproject/zephyr-sdk-0.16.5-1/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32_zephyr-elf/bin/ld.bfd.exe: modules/mbedtls/libmbedTLSCrypto.a(pkparse.c.obj): in function `pk_parse_key_sec1_der':
D:/DEV/zephyrproject/modules/crypto/mbedtls/library/pkparse.c:665: undefined reference to `mbedtls_pk_ecc_set_key'
d:/dev/zephyrproject/zephyr-sdk-0.16.5-1/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32_zephyr-elf/bin/ld.bfd.exe: modules/mbedtls/libmbedTLSCrypto.a(pkparse.c.obj): in function `mbedtls_error_add':
[... cut ...]
D:/DEV/zephyrproject/modules/crypto/mbedtls/library/pkparse.c:421: undefined reference to `mbedtls_pk_ecc_set_pubkey'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'D:\DEV\zephyrproject\zephyr\samples\net\cloud\mqtt_azure\build'

Windows
Zephyr version: 3.6.99 / SDK 0.16.5

Additional context
This is use wifi.  Only fails if overlay conf with wifi added.

Thanks in advance

joelguittet, 2024-05-14:
I confirm the issue, I have the same on another platform.
Seems mbedtls separated files and introduced `MBEDTLS_PK_HAVE_ECC_KEYS` that we don't have in zephyr configuration.
I'm trying to find a workaround, @boaretti have you find something ?

sylvioalves, 2024-05-14:
Folks, would you be able to check this?

sylvioalves, 2024-05-14:
@boaretti I recommend you to also change the following values: 
```c
CONFIG_MBEDTLS_HEAP_SIZE=20480
CONFIG_NET_MGMT_EVENT_STACK_SIZE=4096
```

joelguittet, 2024-05-14:
@sylvioalves you pushed just after my message, I check at your modification for my own use case, this help me too :)

boaretti, 2024-05-14:
Hi all, sadly it didn`t work. Still the same issue. I will post bellow the the confs used for mqtt_azure example ( using it as base ) but other custom project share the same issue when mbedtls added.
prj.conf:

CONFIG_NETWORKING=y
CONFIG_POSIX_API=y

# Disable IPv6 support
CONFIG_NET_IPV6=n

# Enable IPv4 support
CONFIG_NET_IPV4=y
CONFIG_NET_IF_MAX_IPV4_COUNT=2
CONFIG_NET_IF_UNICAST_IPV4_ADDR_COUNT=3

# Enable TCP support
CONFIG_NET_TCP=y

# Enable Sockets support
CONFIG_NET_SOCKETS=y
CONFIG_NET_SOCKETS_SOCKOPT_TLS=y

# Enable DHCPv4 support
CONFIG_NET_DHCPV4=y

# Enable SOCKS5 proxy support
CONFIG_SOCKS=n

# Enable MQTT Lib support
CONFIG_MQTT_LIB=y
CONFIG_MQTT_LIB_TLS=y

# Enable Mbed TLS configuration
CONFIG_MBEDTLS=y
CONFIG_MBEDTLS_BUILTIN=y
CONFIG_MBEDTLS_ENABLE_HEAP=y
CONFIG_MBEDTLS_HEAP_SIZE=20480
CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN=10240
CONFIG_MBEDTLS_PEM_CERTIFICATE_FORMAT=y
CONFIG_NET_MGMT_EVENT_STACK_SIZE=4096
# Network configuration
CONFIG_NET_CONFIG_SETTINGS=y

# Network connection manager
CONFIG_NET_CONNECTION_MANAGER=y

CONFIG_NET_MGMT_EVENT_STACK_SIZE=1024
CONFIG_NET_MGMT_EVENT_QUEUE_SIZE=5
CONFIG_NET_MGMT_EVENT_LOG_LEVEL_DBG=n

CONFIG_DNS_RESOLVER=y
CONFIG_DNS_SERVER_IP_ADDRESSES=y
CONFIG_DNS_SERVER1="8.8.8.8"
CONFIG_DNS_RESOLVER_ADDITIONAL_BUF_CTR=2
CONFIG_REQUIRES_FULL_LIBC=y

CONFIG_INIT_STACKS=y
CONFIG_NET_SHELL=y

CONFIG_MAIN_STACK_SIZE=4096
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=2048

# Enable Logging support
CONFIG_LOG_MODE_IMMEDIATE=y
CONFIG_NET_LOG=y
CONFIG_NET_TCP_LOG_LEVEL_DBG=n
CONFIG_NET_SOCKETS_LOG_LEVEL_DBG=n
CONFIG_MQTT_LOG_LEVEL_DBG=n
CONFIG_NET_DHCPV4_LOG_LEVEL_INF=n
CONFIG_NET_IF_LOG_LEVEL_DBG=n
CONFIG_MBEDTLS_DEBUG=n
CONFIG_SOCKS_LOG_LEVEL_DBG=n
CONFIG_NET_CONFIG_LOG_LEVEL_DBG=y
CONFIG_NET_CONNECTION_MANAGER_LOG_LEVEL_DBG=n
CONFIG_DNS_RESOLVER_LOG_LEVEL_DBG=n
CONFIG_NET_CONTEXT_LOG_LEVEL_DBG=n

esp_procpu.conf:
CONFIG_WIFI=y

CONFIG_NETWORKING=y
CONFIG_NET_L2_ETHERNET=y

CONFIG_NET_IPV6=n
CONFIG_NET_IPV4=y
CONFIG_NET_DHCPV4=y
CONFIG_ESP32_WIFI_STA_AUTO_DHCPV4=y

CONFIG_NET_LOG=y

>>> FUN FACT : It does work fine for other boards like Nucleos !

Thanks All [ ]

sylvioalves, 2024-05-14:
@boaretti What do you mean by it didnt work? I tested here and was able to run azure_mqtt sample code without any issue? Humm, did you apply the PR linked in this issue?

boaretti, 2024-05-15:
@sylvioalves yes, the changes suggested were applied, so the initial issue still despite those changes.

PR #72752 (2024-05-16): drivers: wifi: esp32: remove pre-defined mbedTLS requirements
In #72651, build fails due to conflict when enabling mbedTLS components. Current Wi-Fi implementation for ESP32 can discard those selected cryptos.

Fixes #72651
closed 2024-05-16 by nashif