zephyrproject-rtos/zephyr #72655: zephyr/samples/drivers/watchdog fails on esp32s3_touch_lcd_1_28

nordic-segl, 2024-05-13:
**Describe the bug**
The zephyr/samples/drivers/watchdog fails.

Observed on:
- Board: esp32s3_touch_lcd_1_28, qualifiers: esp32s3/appcpu
- Board: nrf54l15pdk, Revision: 0.2.1, qualifiers: nrf54l15/cpuflpr

**To Reproduce**
Steps to reproduce the behaviour:
1. have esp32s3_touch_lcd_1_28 or nrf54l15pdk connected to PC
2. go to your zephyr directory
3. call ./scripts/twister -T tests/drivers/watchdog -p nrf54l15pdk/nrf54l15/cpuflpr --device-testing --device-serial /dev/ttyACM0 -v --inline-logs
4. See console output with error

**Expected behavior**
Test shall compile and pass.

**Impact**
Other PRs are blocked, f.e. https://github.com/zephyrproject-rtos/zephyr/pull/72566

**Logs and console output**
Most likely, target yaml has "supports watchdog" like here:
https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/nordic/nrf54l15pdk/nrf54l15pdk_nrf54l15_cpuflpr.yaml#L17
But board DTS doesn't define alias watchdog0 used in the test.

On esp32s3_touch_lcd_1_28:

```
Build failure
Loading Zephyr default modules (Zephyr base).
-- Application: /__w/zephyr/zephyr/samples/drivers/watchdog
-- CMake version: 3.22.1
-- Found Python3: /usr/bin/python (found suitable version "3.10.12", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /root/.cache/zephyr
-- Zephyr version: 3.6.99 (/__w/zephyr/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: esp32s3_touch_lcd_1_28, qualifiers: esp32s3/appcpu
-- Found host-tools: zephyr 0.16.5 (/opt/toolchains/zephyr-sdk-0.16.5-1)
-- Found toolchain: zephyr 0.16.5 (/opt/toolchains/zephyr-sdk-0.16.5-1)
-- Found Dtc: /opt/toolchains/zephyr-sdk-0.16.5-1/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /__w/zephyr/zephyr/boards/waveshare/esp32s3_touch_lcd_1_28/esp32s3_touch_lcd_1_28_esp32s3_appcpu.dts
-- Generated zephyr.dts: /__w/zephyr/zephyr/twister-out/esp32s3_touch_lcd_1_28_esp32s3_appcpu/samples/drivers/watchdog/sample.drivers.watchdog/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /__w/zephyr/zephyr/twister-out/esp32s3_touch_lcd_1_28_esp32s3_appcpu/samples/drivers/watchdog/sample.drivers.watchdog/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /__w/zephyr/zephyr/twister-out/esp32s3_touch_lcd_1_28_esp32s3_appcpu/samples/drivers/watchdog/sample.drivers.watchdog/zephyr/dts.cmake
Parsing /__w/zephyr/zephyr/Kconfig
Loaded configuration '/__w/zephyr/zephyr/boards/waveshare/esp32s3_touch_lcd_1_28/esp32s3_touch_lcd_1_28_esp32s3_appcpu_defconfig'
Merged configuration '/__w/zephyr/zephyr/samples/drivers/watchdog/prj.conf'
Merged configuration '/__w/zephyr/zephyr/twister-out/esp32s3_touch_lcd_1_28_esp32s3_appcpu/samples/drivers/watchdog/sample.drivers.watchdog/zephyr/misc/generated/extra_kconfig_options.conf'
Configuration saved to '/__w/zephyr/zephyr/twister-out/esp32s3_touch_lcd_1_28_esp32s3_appcpu/samples/drivers/watchdog/sample.drivers.watchdog/zephyr/.config'
Kconfig header saved to '/__w/zephyr/zephyr/twister-out/esp32s3_touch_lcd_1_28_esp32s3_appcpu/samples/drivers/watchdog/sample.drivers.watchdog/zephyr/include/generated/autoconf.h'
-- Found GnuLd: /opt/toolchains/zephyr-sdk-0.16.5-1/xtensa-espressif_esp32s3_zephyr-elf/xtensa-espressif_esp32s3_zephyr-elf/bin/ld.bfd (found version "2.38") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/toolchains/zephyr-sdk-0.16.5-1/xtensa-espressif_esp32s3_zephyr-elf/bin/xtensa-espressif_esp32s3_zephyr-elf-gcc
esptool path: /__w/zephyr/modules/hal/espressif/tools/esptool_py/esptool.py
-- Using ccache: /usr/local/bin/ccache
CMake Warning at /__w/zephyr/zephyr/CMakeLists.txt:874 (message):
  No SOURCES given to Zephyr library: drivers__watchdog

  Excluding target from build.

-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    TC_RUNID

-- Build files have been written to: /__w/zephyr/zephyr/twister-out/esp32s3_touch_lcd_1_28_esp32s3_appcpu/samples/drivers/watchdog/sample.drivers.watchdog
[... 149 lines cut ...]
/__w/zephyr/zephyr/include/zephyr/device.h:91:41: note: each undeclared identifier is reported only once for each function it appears in
   91 | #define DEVICE_NAME_GET(dev_id) _CONCAT(__device_, dev_id)
      |                                         ^~~~~~~~~
/__w/zephyr/zephyr/include/zephyr/toolchain/common.h:137:26: note: in definition of macro '_DO_CONCAT'
  137 | #define _DO_CONCAT(x, y) x ## y
      |                          ^
/__w/zephyr/zephyr/include/zephyr/device.h:91:33: note: in expansion of macro '_CONCAT'
   91 | #define DEVICE_NAME_GET(dev_id) _CONCAT(__device_, dev_id)
      |                                 ^~~~~~~
/__w/zephyr/zephyr/include/zephyr/device.h:228:37: note: in expansion of macro 'DEVICE_NAME_GET'
  228 | #define DEVICE_DT_NAME_GET(node_id) DEVICE_NAME_GET(Z_DEVICE_DT_DEV_ID(node_id))
      |                                     ^~~~~~~~~~~~~~~
/__w/zephyr/zephyr/include/zephyr/device.h:245:34: note: in expansion of macro 'DEVICE_DT_NAME_GET'
  245 | #define DEVICE_DT_GET(node_id) (&DEVICE_DT_NAME_GET(node_id))
      |                                  ^~~~~~~~~~~~~~~~~~
/__w/zephyr/zephyr/samples/drivers/watchdog/src/main.c:85:42: note: in expansion of macro 'DEVICE_DT_GET'
   85 |         const struct device *const wdt = DEVICE_DT_GET(DT_ALIAS(watchdog0));
      |                                          ^~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.4 LTS
 - zephyr-sdk-0.16.5-1
 - Zephyr version: 3.6.99

sylvioalves, 2024-05-15:
@henrikbrixandersen ESP32 related issue has been  by #72662

nordic-segl, 2024-05-15:
Fix for failures on nrf54l15/cpuflpr is here https://github.com/zephyrproject-rtos/zephyr/pull/72566

aescolar, 2024-05-15:
Closing as both reported boards with failures have had their respective fixes merged.

closed 2024-05-15 by aescolar