zephyrproject-rtos/zephyr #72657: tests: drivers: spi: tests/drivers/spi/dt_spec/drivers.spi.dt_spec and tests/drivers/spi/spi_loopback/drivers.spi.loopback BUS FAULT

nordic-piks, 2024-05-13:
**Describe the bug**
The tests/drivers/spi/spi_loopback/drivers.spi.loopback and tests/drivers/spi/dt_spec/drivers.spi.dt_spec

Observed for
- nrf54h20dk/nrf54h20/cpuapp

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf54h20dk/nrf54h20/cpuapp connected
2. go to your zephyr dir
3. call ./scripts/twister -T tests/drivers/spi -p nrf54h20dk/nrf54h20/cpuapp--device-testing --device-serial /dev/ttyACM0 -v --inline-logs
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
E: ***** BUS FAULT *****
E:   Precise data bus error
E:   BFAR Address: 0x5f09502c
E: r0/a1:  0x00000008  r1/a2:  0x00000000  r2/a3:  0x5f095000
E: r3/a4:  0x0000002c r12/ip:  0x0e0acea1 r14/lr:  0x0e0a8d11
E:  xpsr:  0x69000000
E: Faulting instruction address (r15/pc): 0x0e0aceae
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x2f0112c8 (main)
E: Halting system
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.16.5
 - Commit SHA or Version used: v3.6.0-3753-gbe682e22e166

nordic-piks, 2024-05-13:
@e-rk There seems to be hypothesis that after #71989 this test fails (ie. adding in this test /delete-node/ &exmif; helps)

e-rk, 2024-05-13:
This is because of lack of this commit: https://github.com/nrfconnect/sdk-zephyr/commit/7ea6937f02affb74cae2327aeca955ae5e5c3106
The commit is not suitable for upstream in this form and larger changes to the `spi_dw.c` will be needed.

nordic-piks, 2024-05-14:
The same for tests/drivers/spi/spi_slave/drivers.spi_slave.api for nrf54h20dk/nrf54h20/cpuapp

PR #72928 (2024-05-17): dts: nordic: nrf54h20: Fix exmif node definition
Remove the "snps,designware-spi" compatible from the EXMIF node in nRF54H20i, as the spi_dw driver cannot be used for this peripheral without Nordic-specific modifications that are not present upstream. An attempt to do so (just setting CONFIG_SPI=y will cause that, as the driver initialization function will be executed then) results in a bus fault.

Fixes #72657.
closed 2024-05-17 by MaureenHelm