zephyrproject-rtos/zephyr #72679: Problem with OpenOCD: `board/st_nucleo_l4.cfg`

g40, 2024-05-13:
Using a freshly cloned repo and brand new installation of the Zephyr toolchain as described here: https://docs.zephyrproject.org/latest/develop/getting_started/index.html

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - What have you tried to diagnose or workaround this issue?
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 - ...

**To Reproduce**
Steps to reproduce the behavior:
(As per the getting started section
r:\src\embedded\zephyrproject\zephyr>west build -p always -b nucleo_l476rg samples\basic\blinky

1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
The target board to be flashed via OpenOCD.

**Impact**
Showstopper

**Logs and console output**
```
r:\src\embedded\zephyrproject\zephyr>west flash
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner openocd
-- runners.openocd: Flashing file: r:/src/embedded/zephyrproject/zephyr/build/zephyr/zephyr.hex
Open On-Chip Debugger 0.8.0 (2014-12-03-15:43)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.sourceforge.net/doc/doxygen/bugs.html
Runtime Error: embedded:startup.tcl:47: Can't find board/st_nucleo_l4.cfg
in procedure 'script'
at file "embedded:startup.tcl", line 58
at file "embedded:startup.tcl", line 47
FATAL ERROR: command exited with status 1: R:/apps/bin/openocd.exe -s 'R:/src/embedded/zephyrproject/zephyr/boards/st/nucleo_l476rg\support' -s R:/apps/ztools/zephyr-sdk-0.16.6/sysroots/AMD64-pokysdk-linux/usr/share/openocd/scripts -f 'R:/src/embedded/zephyrproject/zephyr/boards/st/nucleo_l476rg\support\openocd.cfg' '-c init' '-c targets' -c 'reset init' -c 'flash write_image erase r:/src/embedded/zephyrproject/zephyr/build/zephyr/zephyr.hex' -c 'reset run' -c shutdown
```

**Environment (please complete the following information):**
Windows 10x64 22H2
Zephyr SDK
zephyr-sdk-0.16.6

**Additional context**
Targeting an ST Nucleo L_476RG

g40, 2024-05-13:
It's a path problem. See spew below. OpenOCD is on the path, but not according to `west`. Does the tools installation cache paths or something? 

```
r:\src\embedded\zephyrproject\zephyr>west flash
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner openocd
FATAL ERROR: required program R:/apps/bin/openocd.exe not found; install it or add its location to PATH

r:\src\embedded\zephyrproject\zephyr>which openocd
/r/apps/openocd/bin/openocd

r:\src\embedded\zephyrproject\zephyr>where openocd
r:\apps\openocd\bin\openocd.exe

r:\src\embedded\zephyrproject\zephyr>openocd --version
Open On-Chip Debugger 0.12.0 (2023-01-14-23:37)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
```

erwango, 2024-05-13:
Moving to discussions until it is clear there is something to be fixed