zephyrproject-rtos/zephyr #72696: arch: arm: cortex_a_r: XIP=y causes linker error

sigmundklaa, 2024-05-13:
When a Cortex A/R based SoC is compiled with `CONFIG_XIP=y`, linking fails with the following syntax error:
```
/home/sigmundk/src/zephy-extra/zephyr-extra/toolchain/be-gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/9.3.1p1/../../../../arm-none-eabi/bin/ld.bfd:zephyr/linker_zephyr_pre0.cmd:175                                                                  
: syntax error
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

I am compiling for an `ARM Cortex R4F`, using out-of-tree SoC support for the TMS570LS1224 which is linked at the bottom of the issue. The cause seems to be the below line (from `linker_zephyr_pre0.cmd:175`):
```
bss (NOLOAD) : ALIGN_WITH_INPUT ALIGN(_region_min_align)
```
which is the result of:
https://github.com/zephyrproject-rtos/zephyr/blob/d9b3ffa5eafc1f7e6dad0e57f01d9e787cfd0fc5/include/zephyr/arch/arm/cortex_a_r/scripts/linker.ld#L277

From what I can tell, combining `ALIGN_WITH_INPUT` and `ALIGN` is not supported, however I'm not sure of this. For me, replacing
`SECTION_DATA_PROLOGUE` with `SECTION_PROLOGUE` fixed the issue, but I don't know if that is the best solution.

**To Reproduce**
Steps to reproduce the behavior:
1. Select a Cortex A/R based SoC
2. Compile with `CONFIG_XIP=y
3. See error

**Expected behavior**
Successful compilation

**Impact**
Prevents compilation

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04
 - Toolchain Both the latest version of the Zephyr SDK, as well as the toolchain in the repository linked below was tried.
 - Commit SHA or Version used: 23a03b7a0a494e27316fdd7fdb453a0409b99aa9

**Additional context**
My setup, including a patched big endian ARM GCC toolchain, is available at https://github.com/OrbitNTNU/zephyr-hw

cfriedt, 2024-05-14:
@sigmundklaa - can you potentially reproduce this with an in-tree board?

aescolar, 2024-05-14:
CC @dleach02

sigmundklaa, 2024-05-14:
> @sigmundklaa - can you potentially reproduce this with an in-tree board?

Yes ofcourse. I get the same error when compiling for the `nxp/s32z2xxdc2/s32z270/rtu1` board, using:
```bash
west build -p always -b s32z2xxdc2/s32z270/rtu1 samples/hello_world -- -DCONFIG_XIP=y
```

ntavish, 2024-05-27:
I am facing the same problem, trying to compile zephyr for another chip in the family. My toolchain is even older, GCC 7.5.0 (big endian).

ntavish, 2024-05-31:
@sigmundklaa hello, I managed to build the zephyr toolchain (GCC 12.x) for armv7-r big-endian. Please check https://github.com/zephyrproject-rtos/sdk-ng/pull/765

Here's direct link to the download: https://github.com/zephyrproject-rtos/sdk-ng/suites/24251779094/artifacts/1546357536

sigmundklaa, 2024-05-31:
@ntavish Nice! I can check it out later. Did that in any way impact the XIP issue though?

ntavish, 2024-05-31:
I think so, don't see it anymore, although I have several other errors right now and possibly my build stops before that (seems unlikely though).

sigmundklaa, 2024-05-31:
Just tested, same error on GCC 12.2:
[image]

ntavish, 2024-06-28:
I think @sigmundklaa's reasoning is correct and the two cannot be combined, please check the PR I made.

PR #75053 (2024-07-12): toolchain: gcc: fix SECTION_DATA_PROLOGUE for XIP
Current if CONFIG_XIP is enabled, then ALIGN_WITH_INPUT and the align argument to SECTION_DATA_PROLOGUE are set together, and that is not a valid option combination; if ALIGN_WITH_INPUT is present, then the section's LMA alignment is inherited from the previous section, and that cannot be combined with another alignment option.

The background for this line is in the following commit: 6b3c5e8bb2cee171a233638a1b33e0c92121c50c

Additionally, this is the documentation from ld:

(from https://sourceware.org/binutils/docs/ld/Forced-Output-Alignment. [... cut]
closed 2024-07-12 by nashif