zephyrproject-rtos/zephyr #72699: samples: bluetooth: central_iso: unchecked buffer allocation

msmttchr, 2024-05-13:
**Describe the bug**
In `central_iso` sample (function `iso_timer_timeout`), no check on buffer allocation failure is performed.
See code snippet:
https://github.com/zephyrproject-rtos/zephyr/blob/d9b3ffa5eafc1f7e6dad0e57f01d9e787cfd0fc5/samples/bluetooth/central_iso/src/main.c#L62-L68

As a consequence, a system crash is observed when buffer allocation fails.

**To Reproduce**
Not so easy to reproduce and it depends on the buffer allocation policy.
Error observed while running a new platform against Nordic nRF52840 board.

**Expected behavior**
Sample should be robust against buffer allocation failure, e.g. throwing an error or a warning an retrying.

**Impact**
Annoyance, especially when validating new platform prone to have genuine bugs.

**Logs and console output**
Zephyr crash log.

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain Zephyr SDK
 
**Additional context**
N/A

cvinayak, 2024-05-13:
Please provide details on how to build and test in the `To Reproduce` section, and also mention the commit SHA used to build.

> Actually, from quick check, having K_FOREVER as timeout should avoid to get NULL as return value, but indeed this observed.

Lowering the priority of the issue, as steps to reproduce is unclear.

msmttchr, 2024-05-13:
It looks like related/introduced to/by #71697

jori-nordic, 2024-05-14:
Yep. A bunch of (mostly audio or iso-related) samples do this.

Samples should've been programmed better, and not block the whole syswq by allocating with `K_FOREVER` in a callback.
I was not familiar enough with them to propose a patch when I did #71697 . Maybe someone else is?

Note that getting `NULL` was probably possible before:
https://github.com/zephyrproject-rtos/zephyr/pull/71697/files#diff-7d20cda516d057def150f03b4d2cdce79dc768dbc8ec7c3fab9d463265572466L1322-L1325

closed 2024-07-06 by aescolar