zephyrproject-rtos/zephyr #72716: tests: crypto: mbedtls: crypto.mbedtls does not have permission on entropy driver

nordic-piks, 2024-05-14:
**Describe the bug**
The tests/crypto/mbedtls/crypto.mbedtls

Observed for
- nrf52dk/nrf52832
- nrf52840dk/nrf52840
- nrf5340dk/nrf5340/cpunet
- nrf9160dk/nrf9160/ns
- nrf5340dk/nrf5340/cpuapp/ns

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf52840dk/nrf52840 connected
2. go to your zephyr dir
3. call ./scripts/twister -T tests/crypto/mbedtls -p nrf52840dk/nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build v3.6.0-3768-g961959fc5f2b ***
Running TESTSUITE mbedtls_fn
===================================================================
START - test_mbedtls
SHA-256 test #1: passed
SHA-256 test #2: passed
SHA-256 test #3: passed
DES -ECB- 56 (dec): passed
DES -ECB- 56 (enc): passed
DES3-ECB-112 (dec): passed
DES3-ECB-112 (enc): passed
DES3-ECB-168 (dec): passed
DES3-ECB-168 (enc): passed
DES -CBC- 56 (dec): passed
DES -CBC- 56 (enc): passed
DES3-CBC-112 (dec): passed
DES3-CBC-112 (enc): passed
DES3-CBC-168 (dec): passed
DES3-CBC-168 (enc): passed
AES note: built-in implementation.
AES-ECB-128 (dec): passed
AES-ECB-128 (enc): passed
AES-ECB-192 (dec): passed
AES-ECB-192 (enc): passed
AES-ECB-256 (dec): passed
AES-ECB-256 (enc): passed
AES-CBC-128 (dec): passed
AES-CBC-128 (enc): passed
AES-CBC-192 (dec): passed
AES-CBC-192 (enc): passed
AES-CBC-256 (dec): passed
AES-CBC-256 (enc): passed
MPI test #1 (mul_mpi): passed
MPI test #2 (div_mpi): passed
MPI test #3 (exp_mod): passed
MPI test #4 (inv_mod): passed
MPI test #5 (simple gcd): passed
RSA key validation: passed
PKCS#1 encryption : passed
PKCS#1 decryption : passed
CTR_DRBG (PR = TRUE) : passed
CTR_DRBG (PR = FALSE): passed
ENTROPY test: E: thread 0x20004080 (0) does not have permission on entropy driver 0x1380c
E: permission bitmap
E: 00 00                   |..      
E: syscall z_vrfy_device_is_ready failed check: access denied
E: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0x00000000
E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x00000000
E:  xpsr:  0x00000000
E: Faulting instruction address (r15/pc): 0x00000000
E: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
E: Current thread: 0x20004080 (test_mbedtls)
E: Halting system
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.16.5
 - Commit SHA or Version used: v3.6.0-3768-g961959fc5f2b

nordic-piks, 2024-05-14:
@valeriosetti I guess that this failure may be connected to #71827 or #71830

valeriosetti, 2024-05-14:
@nordic-piks thanks for letting me know. I will look into this immediately! Can you please assign this to me (I cannot do that)?

nordic-piks, 2024-05-14:
> @nordic-piks thanks for letting me know. I will look into this immediately! Can you please assign this to me (I cannot do that)?

Unfortunately I cannot make assignment.

aescolar, 2024-05-14:
Duplicate of #72625

closed 2024-05-14 by aescolar