zephyrproject-rtos/zephyr #72741: Settings FILE subsystem is not working with SD card with FAT filesystem

Mierunski, 2024-05-14:
**Describe the bug**
Settings FILE subsystem is not working with SD card with FAT filesystem, providing file directory with `/SD:/` prefix results in fail to load or create settings file.

I'm testing on nRF9160DK with external SD Card module.

When settings file subsystem is trying to create directory for file, it checks directories if they exist one by one. 
Because mount point is first directory in path, `fs_stat` returns `-ENOENT` after that `fs_mkdir` is called which returns `-EEXIST`
https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/settings/src/settings_file.c#L524

For workaround, i tested substitiuting expression in https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/settings/src/settings_file.c#L548 to `err && err != -EEXIST`
After that code is working, but it does not seem like a final solution

**To Reproduce**
Take a board with SD Card, enable Filesystem backend for setting
```
CONFIG_SETTINGS=y
CONFIG_SETTINGS_FILE=y
CONFIG_SETTINGS_FILE_PATH="/SD:/setts.ini"
CONFIG_FILE_SYSTEM=y
CONFIG_FAT_FILESYSTEM_ELM=y
```

**Expected behavior**
Settings file backend works correctly

**Impact**
Settings file backend is useless without additional zephyr patch kept separately

**Logs and console output**
```
*** Booting nRF Connect SDK v3.5.99-ncs1 ***
[00:00:00.577,209] <inf> sd: Maximum SD clock is under 25MHz, using clock of 8000000Hz

Listing dir /SD: ...
[DIR ] System Volume Information
[FILE] setts.ini (size = 486)
[FILE] global_data.json (size = 124)
[DIR ] log
[00:00:00.590,087] <inf> APP: Application started
path = /SD:, err = -2
[00:00:00.590,209] <err> fs: failed to create directory (-17)
settings_subsys_init failed (err -17)
```

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain NCS v2.6.0
 - Version used v3.5.99-ncs1

aescolar, 2024-05-14:
This issue seems to relate to Nordic's Connect SDK.
If the issue is present in Zephyr please provide the information requested in the bug template so the it can be reproduced in upstream Zephyr without NCS.
Otherwise please file the issue downstream instead.

aescolar, 2024-05-14:
@de-nordic @mniestroj

paradajz, 2024-09-04:
Affecting me as well, and can also confirm that adding err && err != -EEXIST condition does indeed fix the issue. Nothing to do with Nordic SDK. Zephyr version v3.7.0

paradajz, 2024-09-11:
Here's the reworked patch I'm using to avoid error on startup, also not sure if it's the actual solution but it works for me - it basically skips calling `mkdir_if_not_exists` if root FatFS directory is detected.

```
diff --git a/subsys/settings/src/settings_file.c b/subsys/settings/src/settings_file.c
index 43dcd1004f5..e75a68c7a21 100644
--- a/subsys/settings/src/settings_file.c
+++ b/subsys/settings/src/settings_file.c
@@ -539,9 +539,15 @@ static int mkdir_for_file(const char *file_path)
 		if (i > 0 && file_path[i] == '/') {
 			dir_path[i] = '\0';
 
-			err = mkdir_if_not_exists(dir_path);
-			if (err) {
-				return err;
+			if (strrchr(dir_path, ':') == &dir_path[strlen(dir_path) - 1]) {
+				LOG_DBG("FatFS root directory detected, skipping mkdir for path: "
+					"%s",
+					dir_path);
+			} else {
+				err = mkdir_if_not_exists(dir_path);
+				if (err) {
+					return err;
+				}
 			}
 		}
```

closed 2025-01-27 by github-actions[bot]