zephyrproject-rtos/zephyr #72763: missed function bt_ctlr_set_public_addr() for EFR32BG22

vasimv, 2024-05-14:
Function bt_ctlr_set_public_addr() is missed when compile for an EFR32BG22 board. It has prototype in header files but no real function (error at linking).

henrikbrixandersen, 2024-05-15:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

jhedberg, 2024-05-17:
Isn't this a native controller API, i.e. it'll only be available on boards that support the native controller? AFAIK EFR32BG22 not such a board.

vasimv, 2024-05-18:
Here is output from west build:

`[291/296] Linking C executable zephyr/zephyr_pre0.elf

FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /home/vasimv/src/zephyr/ibeaconAir/build/zephyr/zephyr_pre0.map 
: && ccache /opt/zephyr-sdk-0.16.6/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -fuse-ld=bfd  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/home/vasimv/src/zephyr/ibeaconAir/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/libarch__arm__core.a  zephyr/arch/arch/arm/core/cortex_m/libarch__arm__core__cortex_m.a  zephyr/arch/arch/arm/core/cortex_m/cmse/libarch__arm__core__cortex_m__cmse.a  zephyr/arch/arch/arm/core/mpu/libarch__arm__core__mpu.a  zephyr/lib/libc/picolibc/liblib__libc__picolibc.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/subsys/random/libsubsys__random.a  zephyr/subsys/bluetooth/common/libsubsys__bluetooth__common.a  zephyr/subsys/bluetooth/host/libsubsys__bluetooth__host.a  zephyr/subsys/net/libsubsys__net.a  zephyr/drivers/adc/libdrivers__adc.a  zephyr/drivers/bluetooth/libdrivers__bluetooth.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/entropy/libdrivers__entropy.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  zephyr/drivers/timer/libdrivers__timer.a  zephyr/drivers/watchdog/libdrivers__watchdog.a  modules/mbedtls/libmbedTLSBase.a  modules/mbedtls/libmbedTLSCrypto.a  modules/mbedtls/libmbedTLSX509.a  modules/mbedtls/libmodules__mbedtls.a  modules/segger/libmodules__segger.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/opt/zephyr-sdk-0.16.6/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/thumb/v8-m.main+fp/hard"  -L/home/vasimv/src/zephyr/ibeaconAir/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  modules/mbedtls/libmbedTLSBase.a  modules/mbedtls/libmbedTLSCrypto.a  modules/mbedtls/libmbedTLSX509.a  -mcpu=cortex-m33  -mthumb  -mabi=aapcs  -mfpu=fpv5-sp-d16  -mfloat-abi=hard  -mfp16-format=ieee  -mtp=soft  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie  -DPICOLIBC_DOUBLE_PRINTF_SCANF  /home/vasimv/src/zephyr/modules/hal/silabs/gecko/../zephyr/blobs/gecko/protocol/bluetooth/bgstack/ll/lib/libbluetooth_controller_efr32xg22_gcc_release.a  /home/vasimv/src/zephyr/modules/hal/silabs/gecko/../zephyr/blobs/gecko/protocol/bluetooth/bgcommon/lib/libbgcommon_efr32xg22_gcc_release.a  /home/vasimv/src/zephyr/modules/hal/silabs/gecko/../zephyr/blobs/gecko/platform/radio/rail_lib/autogen/librail_release/librail_efr32xg22_gcc_release.a  --specs=picolibc.specs  -lc  -lgcc && cd /home/vasimv/src/zephyr/ibeaconAir/build/zephyr && /usr/bin/cmake -E true
/opt/zephyr-sdk-0.16.6/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: app/libapp.a(main.c.obj): in function `set_MAC_addr':
/home/vasimv/src/zephyr/ibeaconAir/src/main.c:343: undefined reference to `bt_ctlr_set_public_addr'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
`

> Isn't this a native controller API, i.e. it'll only be available on boards that support the native controller? AFAIK EFR32BG22 not such a board.

The function is listed in documentation ( https://docs.zephyrproject.org/apidoc/latest/group__bt__ctrl.html ) and has prototype in include/zephyr/bluetooth/controller.h, not sure about its nativeness.

I've made crude hack to change MAC address by replacing bt_id_set_adv_private_addr function to call bt_id_set_adv_random_addr with MAC address i needed and switching to private address in advertisements. Works but not really right way.

> Isn't this a native controller API, i.e. it'll only be available on boards that support the native controller? AFAIK EFR32BG22 not such a board.

jhedberg, 2024-05-18:
@vasimv by "the native controller" I mean what's implemented in `subsys/bluetooth/controller`. AFAIK `controller.h` is an API into that specific subcomponent. The board you're using doesn't support this, rather it has its own controller implementation. The way to interface with other controllers is strictly through HCI, i.e. you'd need normally some vendor HCI command to set the public address (as there's no standard HCI command for it).

jhedberg, 2024-06-12:
Closing as not planned, since this API is only intended to be provided by the native Zephyr controller.

closed 2024-06-12 by jhedberg