zephyrproject-rtos/zephyr #72775: apollo4x: build failure on `kernel/cache/` Twister tests

pillo79, 2024-05-14:
**Describe the bug**

The `kernel/cache` test suite fails on the `apollo4p_evb` and `apollo4p_blue_kxr_evb` boards due to missing symbols. It appears that (the HAL?) does not declare the `SCB_`* cache control functions that are used by `arch/arm/core/cortex_m/cache.c` when `CONFIG_CACHE_MANAGEMENT` is defined. 

**To Reproduce**

Run `west twister -c -T tests/kernel/cache/ -p apollo4p_evb`. 
On current Zephyr main (f5ce0476c5d, https://github.com/zephyrproject-rtos/hal_ambiq/commit/f6858cb3fe3d44945a23d114b7feeeac22ce71cf), tests fail to build with this error:

```
FAILED: zephyr/arch/arch/arm/core/cortex_m/CMakeFiles/arch__arm__core__cortex_m.dir/cache.c.obj 
ccache zephyr-sdk-0.16.4/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DKERNEL -DK_HEAP_MEM_POOL_SIZE=0 -DTC_RUNID=c0f20f410652dfe4df7d454c06f392ed -D__PROGRAM_START -D__ZEPHYR_SUPERVISOR__ -D__ZEPHYR__=1 -Izephyr/kernel/include -Izephyr/arch/arm/include -Izephyr/include -Izephyr/twister-out/apollo4p_evb/tests/kernel/cache/kernel.cache.api.minimallibc/zephyr/include/generated -Izephyr/soc/ambiq -Izephyr/soc/ambiq/apollo4x/. -Izephyr/soc/ambiq/. -Izephyr/subsys/testsuite/include -Izephyr/subsys/testsuite/ztest/include -Icore/include -Imodules/hal/cmsis/CMSIS/Core/Include -Izephyr/modules/cmsis/. -Imodules/hal/ambiq/CMSIS/AmbiqMicro/Include -Imodules/hal/ambiq/utils -Imodules/hal/ambiq/mcu/apollo4p -Imodules/hal/ambiq/mcu/apollo4p/hal -Imodules/hal/ambiq/components/bluetooth -isystem zephyr/lib/libc/minimal/include -isystem zephyr/lib/libc/common/include -isystem zephyr-sdk-0.16.4/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/include -isystem zephyr-sdk-0.16.4/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/include-fixed -Wshadow -fno-strict-aliasing -Werror -Os -imacros zephyr/twister-out/apollo4p_evb/tests/kernel/cache/kernel.cache.api.minimallibc/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfp16-format=ieee --sysroot=zephyr-sdk-0.16.4/arm-zephyr-eabi/arm-zephyr-eabi -imacros zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wdouble-promotion -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=zephyr/tests/kernel/cache=CMAKE_SOURCE_DIR -fmacro-prefix-map=zephyr=ZEPHYR_BASE -fmacro-prefix-map=ZephyrCore=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -MD -MT zephyr/arch/arch/arm/core/cortex_m/CMakeFiles/arch__arm__core__cortex_m.dir/cache.c.obj -MF zephyr/arch/arch/arm/core/cortex_m/CMakeFiles/arch__arm__core__cortex_m.dir/cache.c.obj.d -o zephyr/arch/arch/arm/core/cortex_m/CMakeFiles/arch__arm__core__cortex_m.dir/cache.c.obj -c zephyr/arch/arm/core/cortex_m/cache.c
zephyr/arch/arm/core/cortex_m/cache.c: In function 'arch_dcache_enable':
zephyr/arch/arm/core/cortex_m/cache.c:20:9: error: implicit declaration of function 'SCB_EnableDCache' [-Werror=implicit-function-declaration]
   20 |         SCB_EnableDCache();
      |         ^~~~~~~~~~~~~~~~
```
(and all other `SCB_`* functions).

**Impact**

CI failures ([sample log](https://github.com/zephyrproject-rtos/zephyr/actions/runs/9079517508/job/24949484379?pr=72358)) on recent LLEXT PRs (such as #72358).

RichardSWheatley, 2024-05-14:
I will take a look but we do not define and SCB_ cache functions. I am not sure if those functions map to anything we use.

pillo79, 2024-05-15:
I see. Since we need that Kconfig in llext tests to ensure cache coherency, we will add an exclude to skip the Apollo targets for the time being. Thanks for looking into it!

aescolar, 2024-06-13:
Lowering priority after these boards has been disabled in CI

PR #74275 (2024-06-25): boards: ambiq: apollo4p_evb: fix multiple CI failure
This PR is trying to fix multiple CI failures which were introduced from https://github.com/zephyrproject-rtos/zephyr/pull/70842.
Adds the missing flash controller node in apollo4p_evb to address the CI failure https://github.com/zephyrproject-rtos/zephyr/issues/74212. 
Remove the untestable cache kconfig from apollo4x boards to address the CI failure https://github.com/zephyrproject-rtos/zephyr/issues/72775, https://github.com/zephyrproject-rtos/zephyr/issues/73443 at this moment.
And enable the twister for apollo4x platforms. Refer to http [... cut]
closed 2024-06-25 by nashif