zephyrproject-rtos/zephyr #72790: STM32H743 + more: DMA stream locked when transferring data to/from USART

HelligeChris, 2024-05-15:
Using the UART async driver, when a USART is issuing a DMA request to transfer data, if a concurrent transfer occurs, the requested transfer 
may not be served and the DMA stream may stay locked. This causes the UART async driver to lock up and stop sending packages (uart_tx keeps reporting EBUSY).

Issue is described in the [stm32h7 errata](https://www.st.com/resource/en/errata_sheet/[dm00368411-stm32h742xi-g-and-stm32h743xi-g-device-limitations-stmicroelectronics.pdf](https://www.st.com/resource/en/errata_sheet/dm00368411-stm32h742xi-g-and-stm32h743xi-g-device-limitations-stmicroelectronics.pdf)) section 2.20.6 and effect the following microcontrollers:

* STM32H742xI/G
* STM32H743xI/G
*  STM32H750xB
* STM32H753xI

According to the errata the workaround is to: 
Use the alternative peripheral DMA channel protocol by setting bit 20 of the DMA_SxCR register.
This bit is reserved in the documentation and must be used only on the stream that manages data transfers for 
USART peripherals.

henrikbrixandersen, 2024-05-15:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

HelligeChris, 2024-05-15:
> Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

Hey, I reformatted the issue, though the "Steps to reproduce" section leaves much to be desired...

ajarmouni-st, 2024-05-15:
Hi @HelligeChris

>  if a concurrent transfer occurs,

By this you mean another UART DMA transfer or just another DMA transfer from any other device?

ajarmouni-st, 2024-05-15:
> Hi @HelligeChris
> 
> > if a concurrent transfer occurs,
> 
> By this you mean another UART DMA transfer or just another DMA transfer from any other device?

okey. I just noticed

> we used a DMA to update DAC output at 1MHz

So it's the 2nd case right?

HelligeChris, 2024-05-15:
Hi @ajarmouni-st, from what I can see in the errata it shouldn't matter if it's a UART DMA transfer or DMA transfer to the DAC. Since both the UART and DAC request the DMA to transfer data,  though we only noticed it when using the DMA to update the DAC while the also having the UART running. 

This is properly because the DAC was request DMA transfers so quickly.

aescolar, 2024-07-08:
@ajarmouni-st as there is no assignee, this PR pops up for triaging again. Should be consider this issue resolved?

closed 2024-09-22 by github-actions[bot]

dakejahl, 2025-06-09:
Any updates on this? I am having the same problem in NuttX and simply setting the TRBUFF bit doesn't solve the issue as the errata suggests.