zephyrproject-rtos/zephyr #72793: Kconfig: build system ignores configs for snps/nsim SoCs after HWMv2 was applied

kokas-a, 2024-05-15:
**Describe the bug**
After HWMv2 model was applied, build system ignores ``` config XIP ``` that configured on SoC level of Kconfig.

Synopsys nSIM platforms has following levels of XIP configs:
1. arch/Kconfig applied for ARC architecture in general via ``` imply XIP ```
2. soc/snps/nsim/Kconfig.defconfig via ``` default n ```
3. (optional) board/snps/nsim/ ${board}_defconfig

If XIP configuration is not set on board level explicitly, then 2nd level (SoC-level) config is ignored. And final configuration is taken 
from arch-level with ``` CONFIG_XIP=y ``` value.

**To Reproduce**
Steps to reproduce the behavior:
1. ``` west build -b nsim/nsim_em samples/hello_world ```
2. ``` cat build/zephyr/.config | grep XIP ```
3. See incorrect behavior: ``` CONFIG_XIP=y ```

**Expected behavior**
``` CONFIG_XIP=n ```  in .config file, as config XIP was not explicitly defined on board level. It defined with ``` n ``` value on SoC level.

**Environment (please complete the following information):**
 - OS: LInux
 - Toolchain: Zephyr SDK, ARCMWDT
 - Commit SHA ff58bd02047cc1207e37ca988eb1a0e03a98342e or just main

nashif, 2024-06-04:
@kokas-a  is this resolved now with #73433

kokas-a, 2024-06-05:
Hi @nashif ,
https://github.com/zephyrproject-rtos/zephyr/pull/73433 provides workaround as a temporary solution, but it does not fixes root cause itself. Additionally, issues with the risc-v platform have not yet been resolved (see my additional comment here).

evgeniy-paltsev, 2024-06-05:
@nashif I've updated the issue name so it would be more clear.

It's not ARC-related (or architecture related) issue, but a regression in Kconfig configuration system introduced with HW model v2.

It seems to be that we now handle the `imply` keyword incorrectly - so if we `imply` some Kconfig symbol in kernel Kconfig we can't override it in SoCs-level `Kconfig.defconfig` file. It causes some symbol to be enabled even if we disable it in on SoC level.

For this particular symbol (`CONFIG_XIP`) we've found ARC and RISC-V SoC which configuration was affected by this issue, however I expect that there is more issues with other symbols.

nono313, 2024-09-03:
When building blinky sample for nucleo_h723zg, I also have `CONFIG_XIP=y` though neither the sample nor the `nucleo_h723zg_defconfig` defines it.

nordicjm, 2024-11-12:
I don't see a bug here, the default is correctly set to n, in the arch level you have a Kconfig that **implies** it, imply is not default, imply takes prescience over default and is on par with a select statement.

nordicjm, 2024-11-13:
Closing since not a bug

closed 2024-11-13 by nordicjm