zephyrproject-rtos/zephyr #72798: net: buf: Make a `net_buf_move(&buf)` API

jori-nordic, 2024-05-15:
We sometimes do an implicit move of netbufs.
In that case, the best we do is leave a comment, and maybe `NULL` out the original variable, like this

```c
	/* Move `buf` into `temp` */
	temp = buf;
	buf = NULL;
```

It would be nice to have a proper API to do this in `net/buf.h`, something like
```c
	temp = net_buf_move(&buf);
	__ASSERT_NO_MSG(buf == NULL);
```

It could be defined like so
```c
struct net_buf *net_buf_move(struct net_buf **original)
{
	struct net_buf *new = *original;

	__ASSERT_NO_MSG(new->ref == 1);
	*original = NULL;

	return new;
}
```

jukkar, 2024-05-17:
To me this new function gives impression of moving the data from/to the net_buf, like what `memmove()` does. Not sure what it could be called, but some examples could be `net_buf_replace()`, `net_buf_swap()` or similar.

jhedberg, 2024-05-17:
> To me this new function gives impression of moving the data from/to the net_buf, like what `memmove()` does. Not sure what it could be called, but some examples could be `net_buf_replace()`, `net_buf_swap()` or similar.

@jukkar point taken. That said, we do tend to use `net_buf_*_mem()` in the net_buf API for memory operations. Also, "move" is verb that's used for analogous things in other languages, like Rust. Would it help to append _ref() to the end of the API, i.e. `net_buf_move_ref()`? (not sure I prefer that more, but that could just be due to my dislike for unnecessarily long symbol names :)

jukkar, 2024-05-17:
Yes, `net_buf_move_ref()` would indeed be better here.