zephyrproject-rtos/zephyr #72807: Lots of twister errors in CI that don't get reported as failures

arbrauns, 2024-05-15:
There are a lot of python exceptions being thrown in twister CI runs, but the run doesn't get marked as failed:

```
Process Process-17:
Traceback (most recent call last):
  File "/usr/lib/python3.10/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/usr/lib/python3.10/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/__w/zephyr/zephyr/scripts/pylib/twister/twisterlib/runner.py", line [132](https://github.com/zephyrproject-rtos/zephyr/actions/runs/9094695276/job/24996411498#step:11:134)5, in pipeline_mgr
    pb.process(pipeline, done_queue, task, lock, results)
  File "/__w/zephyr/zephyr/scripts/pylib/twister/twisterlib/runner.py", line 677, in process
    self.run()
  File "/__w/zephyr/zephyr/scripts/pylib/twister/twisterlib/runner.py", line 1[134](https://github.com/zephyrproject-rtos/zephyr/actions/runs/9094695276/job/24996411498#step:11:136), in run
    instance.handler.handle(harness)
  File "/__w/zephyr/zephyr/scripts/pylib/twister/twisterlib/handlers.py", line 306, in handle
    with subprocess.Popen(command, stdout=subprocess.PIPE,
  File "/usr/lib/python3.10/subprocess.py", line 971, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.10/subprocess.py", line 1863, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/__w/zephyr/zephyr/twister-out/unit_testing/tests/unit/winstream/utilities.winstream/zephyr/zephyr.exe'
```

Example: https://github.com/zephyrproject-rtos/zephyr/actions/runs/9094695276/job/24996411498#step:11:131

As far as I can tell, this kills the twister process and half the tests aren't even being run, yet no failure is reported in the GitHub action.

PerMac, 2024-05-16:
I found the root cause of this. Will provide a fix soon

nashif, 2024-05-16:
> the above link fixes "unit" type tests. Next PR will fix the exception handling so such errors are correctly reported

what next PR? where is it?

PerMac, 2024-05-16:
WIP. Probably won't be able to push it today. The error with failing test is fixed. IMO failing unit tests were high priority. However, the fact that twister was not able to properly handle those issue is not that high, especially that this was already like this and the recent unit test errors just exposed this

nashif, 2024-05-16:
ok, I will try and push something to address this

PerMac, 2024-05-16:
I just opened a PR with proposition

PR #72893 (2024-06-04): twister: handle exceptions when running binaries
Deal with binaries not found. Also set default return code to something
other than zero to catch issues and report correct status.

Fixes #72807
commit d877d299 (2024-06-04): twister: abort on execution and retun 0 on exceptions

When something goes wrong with execution, due to twister issues and
bugs, do not continue with execution and abort the process and report
and return an error code.

Fixes #72807
closed 2024-06-04 by henrikbrixandersen