zephyrproject-rtos/zephyr #72818: Runtime Filtering of Log Levels Not Working Properly

satheeshkola, 2024-05-15:
**Bug Found: Issue with Runtime Log Level Filtering in Zephyr**

**Description:**  I've encountered an issue regarding the runtime filtering of log levels in Zephyr. Despite configuring the project with 
`LOG_MODULE_REGISTER(MONITOR)` 

and attempting to set the log filter using api:
`log_filter_set(NULL, 0, log_source_id_get(STRINGIFY(MONITOR)), LOG_LEVEL_WRN), `

I've noticed that the filtering doesn't apply as expected when the backend parameter is set to null for the module.

**Project Configuration Summary:**
    • 12 frontends and 1 backend are utilized.
    • Runtime filtering is enabled with `CONFIG_LOG_RUNTIME_FILTERING=y.`
    • UART backend is disabled with `CONFIG_LOG_BACKEND_UART=n.`

**Expected Behavior:** Setting the log level filter using log_filter_set at runtime should effectively control the log output for the specified module.

**Actual Behavior:** Despite configuring the requested log level here in above eg: `LOG_LEVEL_WRN` using log_filter_set, the filtering doesn't function as anticipated when the backend parameter is set to null.

**Steps to Reproduce:**
    1. Configure the project as described above.
    2. Attempt to set log filtering using log_filter_set at runtime.
    3. Observe that the filtering does not behave as expected when the backend parameter is null.
    
    Environment:
    Linux,
    zephyr-sdk(zephyr 0.16.5)
   
    
I kindly request assistance in resolving this issue. Thank you.

nordic-krch, 2024-05-16:
Thanks for submitting the issue. Can you clarify few things?
>  12 frontends and 1 backend are utilized.

What do you mean by 12 frontends? Logging support only one user frontend (`log_frontend.h` API implementation).
What kind of backend do you use? When is it started (from the beginning or later during runtime?). What is the unexpected behavior (`INF` level logs are still present vs no logs)? We are using this in `samples/subsys/logging/logger` and it works correctly there.

satheeshkola, 2024-05-16:
1.I have registered modules, I have registered 12 LOG modules.
2. Custom backend using UART.
3. Custom backed is start from the beginning and trying to change Log level at runtime.
4. Even though i change the log level to WARN still i am getting INF logs.

closed 2024-08-04 by github-actions[bot]