zephyrproject-rtos/zephyr #72835: Fix the temperature ADC_STM32_Die_Temp_Read value getting incorrect for nucleo_h743zi in Zephyr 3.6.0

ShubhadaHarkut-eaton, 2024-05-16:
nucleo_h743zi: incorrect ADC_STM32_Die_Temp_Read temperature value.
Zephyr 3.2.0

nucleo_h743zi.overlay
/{
zephyr,user {
/* adjust channel number according to pinmux in board.dts */

	io-channels = <&adc3 18>;

};
};

...
To Reproduce
Steps to reproduce the behavior:
To Build: west build -b nucleo_h743zi ./ --pristine
To Flash: west flash
4. See error
*** Booting Zephyr OS build v3.6.0
STM32 Die temperature sensor test
Current temperature: -173.3 �°C
Current temperature: -126.2 �°C
Current temperature: -128.2 �°C
Current temperature: -128.8 �°C
Current temperature: -128.8 �°C
Current temperature: -126.9 �°C
Current temperature: -126.2 �°C
Current temperature: -126.5 �°C
Current temperature: -126.9 �°C
Expected behavior
positive value

Impact
High

Logs and console output

STM32 Die temperature sensor test
Current temperature: -173.3 �°C
Current temperature: -126.2 �°C
Current temperature: -128.2 �°C
Current temperature: -128.8 �°C
Current temperature: -128.8 �°C
Current temperature: -126.9 �°C
Current temperature: -126.2 �°C
Current temperature: -126.5 �°C
Current temperature: -126.9 �°C

Environment (please complete the following information):

OS: zephyr-v3.2.0
Toolchain zephyr-sdk-0.15.2
Commit SHA or Version used

FRASTM, 2024-05-21:
@ShubhadaHarkut-eaton can you please check the boards DTS. 
With sha1   we see a correct value:
west build -p -b nucleo_h743zi  
```
*** Booting Zephyr OS build v3.6.0-4280-g105fc792675c ***
CPU Die temperature[dietemp]: 28.4 °C
```

gautierg-st, 2024-06-26:
Hello @ShubhadaHarkut-eaton,
I've just pushed #75048 that could solve the measurement problem you're seeing.
Would it be possible for you to test and let me know?
Thanks

erwango, 2024-06-27:
@ShubhadaHarkut-eaton Zephyr release V3.2.0 is not maintained anymore (see https://github.com/zephyrproject-rtos/zephyr/wiki/Release-Management).
Hence I'm closing this point. ,

But please note above comment, as feedback on #75048 would be helpful.

FRASTM, 2024-07-17:
closing the issue as PR is now merged

closed 2024-07-17 by FRASTM