zephyrproject-rtos/zephyr #72840: sensor_shell: strange and unexpected output when getting single axis from 3-axis data

ycsin, 2024-05-16:
**Describe the bug**
Getting the data of a single axis from 3-axis data is kinda weird, the following output is from #72833:

```
uart:~$ sensor get sensor@0 0
channel idx=3 accel_xyz shift=0 num_samples=0 value=4294967295ns, (-0.000000, -0.000000, -0.000000)
[870:26:48.134,000] <inf> fake_sensor: [sample_fetch] dev: 0x8000e248, chan: 59
[870:26:48.134,000] <inf> fake_sensor: [channel_get] dev: 0x8000e248, chan: 0
uart:~$ sensor get sensor@0 1
[870:26:48.134,000] <inf> fake_sensor: [sample_fetch] dev: 0x8000e248, chan: 59
[870:26:48.134,000] <inf> fake_sensor: [channel_get] dev: 0x8000e248, chan: 1
uart:~$ sensor get sensor@0 2
[870:26:48.134,000] <inf> fake_sensor: [sample_fetch] dev: 0x8000e248, chan: 59
[870:26:48.134,000] <inf> fake_sensor: [channel_get] dev: 0x8000e248, chan: 2
uart:~$ sensor get sensor@0 3
channel idx=3 accel_xyz shift=31 num_samples=1 value=9220133425350000000ns, (3.000000, 1.000000, -391.000000)
[870:26:48.134,000] <inf> fake_sensor: [sample_fetch] dev: 0x8000e248, chan: 59
[870:26:48.134,000] <inf> fake_sensor: [channel_get] dev: 0x8000e248, chan: 3
uart:~$ sensor get sensor@0 4
channel idx=7 gyro_xyz shift=0 num_samples=0 value=4294967295ns, (-0.000000, -0.000000, -0.000000)
[870:26:48.134,000] <inf> fake_sensor: [sample_fetch] dev: 0x8000e248, chan: 59
[870:26:48.134,000] <inf> fake_sensor: [channel_get] dev: 0x8000e248, chan: 4
uart:~$ sensor get sensor@0 5
[870:26:48.134,000] <inf> fake_sensor: [sample_fetch] dev: 0x8000e248, chan: 59
[870:26:48.134,000] <inf> fake_sensor: [channel_get] dev: 0x8000e248, chan: 5
uart:~$ sensor get sensor@0 6
[870:26:48.134,000] <inf> fake_sensor: [sample_fetch] dev: 0x8000e248, chan: 59
[870:26:48.134,000] <inf> fake_sensor: [channel_get] dev: 0x8000e248, chan: 6
uart:~$ sensor get sensor@0 7
channel idx=7 gyro_xyz shift=31 num_samples=1 value=9220133425350000000ns, (7.000000, 1.000000, -391.000000)
[870:26:48.134,000] <inf> fake_sensor: [sample_fetch] dev: 0x8000e248, chan: 59
[870:26:48.134,000] <inf> fake_sensor: [channel_get] dev: 0x8000e248, chan: 7
uart:~$ sensor get sensor@0 37
channel idx=37 pos_dx shift=0 num_samples=0 value=4294967295ns, (-0.000000, -0.000000, -0.000000)
[870:26:48.134,000] <inf> fake_sensor: [sample_fetch] dev: 0x8000e248, chan: 59
[870:26:48.134,000] <inf> fake_sensor: [channel_get] dev: 0x8000e248, chan: 37
uart:~$ sensor get sensor@0 38
[870:26:48.134,000] <inf> fake_sensor: [sample_fetch] dev: 0x8000e248, chan: 59
[870:26:48.134,000] <inf> fake_sensor: [channel_get] dev: 0x8000e248, chan: 38
uart:~$ sensor get sensor@0 39
[870:26:48.134,000] <inf> fake_sensor: [sample_fetch] dev: 0x8000e248, chan: 59
[870:26:48.134,000] <inf> fake_sensor: [channel_get] dev: 0x8000e248, chan: 39
uart:~$ sensor get sensor@0 40
channel idx=40 rpm shift=6 num_samples=1 value=9220133425350000000ns (40.999999)
[870:26:48.134,000] <inf> fake_sensor: [sample_fetch] dev: 0x8000e248, chan: 59
[870:26:48.134,000] <inf> fake_sensor: [channel_get] dev: 0x8000e248, chan: 40
```

i.e., there's no output for channel(s):
- 1 (acc - y)
- 2 (acc - z)
- 5 (gyro - y)
- 6 (gyro - z)
- ...

**To Reproduce**
Steps to reproduce the behavior:
1. Build the sensor_shell sensor with board that has a acc/mag/gyro, or check-out #72833
2. Use `sensor get` for the y-axis

**Expected behavior**
Use should be able to get any supported channel from the sensor

**Impact**
`sensor get` is not able to sample some of the channel(s).

**Logs and console output**
See above

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: 0.16.5-1
 - Upstream `main`

teburd, 2024-05-16:
@yperess please take a look when you can

ubieda, 2024-07-03:
I believe this issue is not being tackled right now. Would you like me to take a look?

teburd, 2024-07-16:
> I believe this issue is not being tackled right now. Would you like me to take a look?

Please do if you have the time

SzymonRichert, 2024-08-27:
@ubieda can you help ?

PR #83474 (2025-01-09): sensor: shell: Allow individual axis data to be processed
### Description
This PR allows the sensor-shell to process individual axis data by making two changes:
1. Allow _sensor_shell_ to request/process individual axis data.
2. Fix the default decoder to properly distinguish individual axis data from 3-axis data.

Fixes #72840

### Testing

- Build and Run
``` console
west build -b native_sim -t run samples/sensor/sensor_shell/ -- -DCONFIG_SAMPLES_SENSOR_SHELL_FAKE_SENSOR=y
```
- Console output
``` console
uart:~$ sensor get sensor@0 accel_x
channel type=0(accel_x) index=0 shift=0 num_samples [... cut]
PR #83723 (2025-01-23): [Backport v3.7-branch] sensor: shell: Allow individual axis data to be processed
Backport 1e64d7bea1d45d580e62cde661408d597ade4331~4..1e64d7bea1d45d580e62cde661408d597ade4331 from #83474.

Fixes #72840
PR #83724 (2025-02-04): [Backport v4.0-branch] sensor: shell: Allow individual axis data to be processed
Backport 1e64d7bea1d45d580e62cde661408d597ade4331~4..1e64d7bea1d45d580e62cde661408d597ade4331 from #83474.

Fixes #72840