zephyrproject-rtos/zephyr #72845: Errors in rx node retention mechanism in refactored LLCP, possibly leading to erroneous rx node handling

erbr-ot, 2024-05-16:
Two rx node specific errors were identified in the refactored LLCP implementation. Related to the rx node retention mechanism.
1. In case a remote CPR is initiated while one is ongoing on other connection this would under specific circumstances lead to an rx node being retained, but not released. 
2. In case of a remote channel map update procedure the rx node reference gets stored but not cleared again, which will lead to possible error on the completion (at instant) - IF the rx node referenced has since been re-allocated for use in another procedure on other connection AND this has then been marked for retention, THEN the completion of channel map update, would see a non-released, retained rx node and then mark it for release. Possibly resulting in an assertion in the parallel procedure.

These issues are both present in refactored LLCP.

The result of these issues is that rx nodes can leak or be improperly released/modified. Consequently resulting in an assertion and/or 'un-explainable' rx node corruption.

erbr-ot, 2024-05-16:
fixed by: https://github.com/zephyrproject-rtos/zephyr/pull/71934

henrikbrixandersen, 2024-05-16:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

nashif, 2024-06-04:
@cvinayak  update?

cvinayak, 2024-06-18:
@erbr-ot is this fixed by #71934 ?

erbr-ot, 2024-06-18:
Yes, as stated

closed 2024-06-18 by cvinayak