zephyrproject-rtos/zephyr #72851: drivers: uart: uart_pm: drivers.uart.pm fails

katgiadla, 2024-05-16:
**Describe the bug**
The test `drivers/uart/uart_pm/drivers.uart.pm` fails.

Observed for `nrf54l15pdk/nrf54l15/cpuapp`.

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf54l15pdk/nrf54l15/cpuapp` connected
2. go to your zephyr dir
3. call `./scripts/twister -T tests/drivers/uart/uart_pm -p nrf54l15pdk/nrf54l15/cpuapp --device-testing --device-serial /dev/ttyACM0 -v --inline-logs --fixture gpio_loopback`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build v3.6.0-3947-g859cc2f1feec ***
Running TESTSUITE uart_pm
===================================================================
START - test_uart_pm_in_idle
Assertion failed at ../../../../../../../zephyr/tests/drivers/uart/uart_pm/src/main.c:33: polling_verify: err not equal to -1
FAIL - test_uart_pm_in_idle in 0.033 seconds
===================================================================
START - test_uart_pm_poll_tx
Assertion failed at ../../../../../../../zephyr/tests/drivers/uart/uart_pm/src/main.c:33: polling_verify: err not equal to -1
FAIL - test_uart_pm_poll_tx in 0.012 seconds
===================================================================
START - test_uart_pm_poll_tx_interrupted
Assertion failed at ../../../../../../../zephyr/tests/drivers/uart/uart_pm/src/main.c:139: action_run: (err not equal to exp_err)
Unexpected err: -120
ASSERTION FAIL [0] @ ../../../../../../../zephyr/kernel/sched.c:1392
aborting essential thread 0x20000458
E: r0/a1:  0x00000004  r1/a2:  0x00000570  r2/a3:  0x200006f4
E: r3/a4:  0x00000004 r12/ip:  0x200006f4 r14/lr:  0x000075ab
E:  xpsr:  0x21000000
E: Faulting instruction address (r15/pc): 0x000084e6
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x20000458 (sysworkq)
E: Halting system
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain: Zephyr SDK 0.16.5-1
 - Commit SHA or Version used: `v3.6.0-3947-g859cc2f1feec`

**Additional context**
Used pin set:
```
psels = <NRF_PSEL(UART_TX, 1, 10)>,
	<NRF_PSEL(UART_RX, 1, 11)>;
```

nordic-krch, 2024-08-07:
It should be fixed by #76330 (and it works on my setup). @katgiadla Can you check?

katgiadla, 2024-08-07:
yes, it works and the bug report could be closed as fixed.

closed 2024-08-07 by katgiadla