zephyrproject-rtos/zephyr #72852: Bluetooth: Controller: Assert when starting periodic advertising right after extended advertising when there is a BIG

Thalley, 2024-05-16:
**Describe the bug**
If there is a BIG created and periodic advertising is started right after extended advertising, there is an assert in the controller that is triggered: 

```
ASSERTION FAIL [id != ((uint8_t)((uint8_t)0 - 1))] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/ull_adv_aux.c:3218
@ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/ull_adv_aux.c:3218
```

**To Reproduce**
Can be reproduced with https://github.com/zephyrproject-rtos/zephyr/pull/72690

Steps to reproduce the behavior:
1. Remove the k_sleep in `start_ext_adv` from `tests/bsim/bluetooth/host/iso/bis/src/bis_broadcaster.c`
2. Build and run `tests/bsim/bluetooth/host/iso/bis/tests_scripts/bis.sh`
3. See the above assert

**Expected behavior**
Expect the controller to support starting periodic advertising directly after extended advertising. 

**Impact**
Big annoyance, but can be circumvented with a single k_sleep

**Logs and console output**
See above

**Environment (please complete the following information):**
 - Commit SHA or Version used https://github.com/zephyrproject-rtos/zephyr/pull/72690 at 59212c198cf60f5c1380bb4a21eb9ff059ed4f85

**Additional context**
N/A

jori-nordic, 2024-08-05:
Note for would-be issue fixers: Don't forget to grep the zephyr tree for mentions of this issue in order to remove workarounds.
E.g. `tests/bsim/bluetooth/host/iso/bis/src/bis_broadcaster.c` and maybe others.

Thalley, 2025-01-17:
Will rerun to see if this is still an issue

Thalley, 2025-01-17:
The line is slightly changed, but still valid: 
```
d_00: @00:00:00.013337  ASSERTION FAIL [id != ((uint8_t)((uint8_t)0 - 1))] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/ull_adv_aux.c:3250
d_00: @00:00:00.013337  @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/ull_adv_aux.c:3250
d_00: @00:00:00.013337  [00:00:00.013,336] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
d_00: @00:00:00.013337  [00:00:00.013,336] <err> os: Current thread: 0x820fec0 (unknown)
d_00: @00:00:00.013337  [00:00:00.013,336] <err> os: Halting system
d_00: @00:00:00.013337 ERROR: Exiting due to fatal error
```

Thalley, 2025-03-19:
Still valid

closed 2025-07-31 by Thalley