zephyrproject-rtos/zephyr #72887: net: http: HTTP server send incomplete response header

mustafaabdullahk, 2024-05-16:
**Describe the bug**

I have one problem about http response header. When i was trying to last changes to http server, i face to response header length problem. i did save the screen following issue:

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b stm32h747i_disco/stm32h747xx/m7 samples/net/sockets/http_server

**Expected behavior**

[image]

**Impact**

HTTP request finish by fail (error encoding chunk).

**Logs and console output**

```
[00:00:10.071,000] <dbg> net_http_server: accept_new_client: New client from 10.5.190.235:56328
[00:00:10.071,000] <dbg> net_http_server: http_server_run: Init client #0
[00:00:10.071,000] <dbg> net_http_server: handle_http_preface: HTTP_SERVER_PREFACE_STATE.
[00:00:10.071,000] <dbg> net_http_server: handle_http1_request: HTTP_SERVER_REQUEST
[00:00:10.072,000] <dbg> net_http_server: on_header_value: Header value too long (by 68 bytes)
[00:00:10.072,000] <dbg> net_http_server: handle_http1_request: HTTP_SERVER_REQUEST
[00:00:10.072,000] <dbg> net_http_server: on_header_value: Header value too long (by 102 bytes)
[00:00:10.072,000] <dbg> net_http_server: handle_http1_request: Requested URL: /metrics
```

**Environment (please complete the following information):**
 - OS:Linux
 - Toolchain Zephyr SDK
 - [net: http_server: Add support for specifying Content-Type](https://github.com/zephyrproject-rtos/zephyr/pull/72629)

**Additional context**

that block maybe contain strlen bug:

[image]

jukkar, 2024-05-29:
@mustafaabdullahk Is there an easy way to replicate the issue? To me it looks like we might run out of network buffers at some point. Are you running plain http_sample as you seem to have the prometheus support enabled. What does your prj.conf file look like?

mustafaabdullahk, 2024-05-29:
@jukkar Can be used mustafaabdullahk:mustafaabdullahk/prometheus-library branch directly. I did use samples/net/sockets/http_server prj.conf just one diff.

```
# General config
CONFIG_MAIN_STACK_SIZE=3072
CONFIG_SHELL=y
CONFIG_LOG=y
CONFIG_ENTROPY_GENERATOR=y
CONFIG_TEST_RANDOM_GENERATOR=y
CONFIG_INIT_STACKS=y
CONFIG_POSIX_MAX_FDS=32
CONFIG_POSIX_API=y
CONFIG_FDTABLE=y
CONFIG_NET_SOCKETS_POLL_MAX=32
CONFIG_REQUIRES_FULL_LIBC=y
CONFIG_HEAP_MEM_POOL_SIZE=2048

# Prometheus
CONFIG_PROMETHEUS=y

# Eventfd
CONFIG_EVENTFD=y

# Networking config
CONFIG_NETWORKING=y
CONFIG_NET_IPV4=y
CONFIG_NET_IPV6=y
CONFIG_NET_TCP=y
CONFIG_NET_SOCKETS=y
CONFIG_NET_CONNECTION_MANAGER=y
CONFIG_NET_SHELL=y
CONFIG_NET_LOG=y
CONFIG_NET_DHCPV4=y

# JSON
CONFIG_JSON_LIBRARY=y

# HTTP parser
CONFIG_HTTP_PARSER_URL=y
CONFIG_HTTP_PARSER=y
CONFIG_HTTP_SERVER=y
CONFIG_HTTP_SERVER_WEBSOCKET=y
CONFIG_HTTP_SERVER_MAX_CONTENT_TYPE_LENGTH=128

# Network buffers
CONFIG_NET_PKT_RX_COUNT=16
CONFIG_NET_PKT_TX_COUNT=16
CONFIG_NET_BUF_RX_COUNT=128
CONFIG_NET_BUF_TX_COUNT=128
CONFIG_NET_CONTEXT_NET_PKT_POOL=y

# IP address options
CONFIG_NET_IF_UNICAST_IPV6_ADDR_COUNT=3
CONFIG_NET_IF_MCAST_IPV6_ADDR_COUNT=4
CONFIG_NET_MAX_CONTEXTS=32
CONFIG_NET_MAX_CONN=32

# Network address config
CONFIG_NET_CONFIG_SETTINGS=y
CONFIG_NET_CONFIG_NEED_IPV4=y
CONFIG_NET_CONFIG_NEED_IPV6=y
CONFIG_NET_CONFIG_MY_IPV4_ADDR="192.0.2.1"
CONFIG_NET_CONFIG_PEER_IPV4_ADDR="192.0.2.2"
CONFIG_NET_CONFIG_MY_IPV4_GW="192.0.2.2"
CONFIG_NET_CONFIG_MY_IPV6_ADDR="2001:db8::1"
CONFIG_NET_CONFIG_PEER_IPV6_ADDR="2001:db8::2"

# TLS configuration
CONFIG_MBEDTLS=y
CONFIG_MBEDTLS_BUILTIN=y
CONFIG_MBEDTLS_ENABLE_HEAP=y
CONFIG_MBEDTLS_HEAP_SIZE=60000
CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN=2048
CONFIG_NET_SOCKETS_SOCKOPT_TLS=y
CONFIG_NET_SOCKETS_TLS_MAX_CONTEXTS=6
CONFIG_TLS_CREDENTIALS=y
CONFIG_TLS_MAX_CREDENTIALS_NUMBER=5

# Networking tweaks
# Required to handle large number of consecutive connections,
# e.g. when testing with ApacheBench.
CONFIG_NET_TCP_TIME_WAIT_DELAY=0
```

I did just CONFIG_PROMETHEUS.

mustafaabdullahk, 2024-05-30:
Can you see following content:

[image]

my curl output:

[image]

excepted output:

```http_request_counter{http_request="request_count"} 6```

wireshark packet detail:

[image]

last git activities:

[image]

jukkar, 2024-05-30:
Thanks, I could replicate the issue, investigating...

mustafaabdullahk, 2024-05-30:
Your welcome, i apply some patch for solve issue and working. If you want to i create pull request for that, I can.

jukkar, 2024-05-30:
Thanks, I fixed it before I saw your offer. Anyway, please try the PR and see if it works for you.

PR #73516 (2024-06-03): net: http_server: Send chunked response correctly
The chunked response was not sent properly. There were extra "\r\n" before the chunk lenght and the length of the string to be sent was calculated incorrectly.

Fixes #72887
commit a124dd25 (2024-06-03): net: http_server: Send chunked response correctly

The chunked response was not sent properly. There were extra
"\r\n" before the chunk lenght and the length of the string
to be sent was calculated incorrectly.

Fixes #72887
closed 2024-06-03 by aescolar